mit-han-lab/bevfusion #203: question about training process

hualuojingye, 2022-10-27:
Hello,thanks for your contribution in fusion. I'm a freshmen in this area.Recently I'm doing some related work.I have a question about the training process.
Do lidar branch and image branch train together or separately?Looking forward to hearing from you.

VeeranjaneyuluToka, 2022-10-27:
LiDAR barch and Image branch are parallel till the feature extraction stage, then fuse the features and then it is a single pipeline from there. So i think we can say that training happens togather in case of LiDAR and Camera fusion. However would also interested to know if there is a intitutive explanation. Thanks!

zhijian-liu, 2022-10-29:
We first pre-train the LiDAR branch and then fine-tune both branches together.

kentang-mit, 2022-11-10:
Closed due to inactivity.

closed 2022-11-10 by kentang-mit