mit-han-lab/bevfusion #204: some questions about bev_pool

Z-chocking, 2022-10-27:
Hi,thanks for sharing your great work.
Recently i have seen the `bev_pool` and  wanted to know the principle and details,here I have some questions about  it .
You do the similar job as [LSS](https://github.com/nv-tlabs/lift-splat-shoot/blob/d74598cb51101e2143097ab270726a561f81f8fd/src/models.py#L200) in [bev_pool](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/mmdet3d/models/vtransforms/base.py#L137),but in [QuickCumsumCuda](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/mmdet3d/ops/bev_pool/bev_pool.py#L37),you compute the `kept` in the way which differents from LSS:
In [LSS](https://github.com/nv-tlabs/lift-splat-shoot/blob/d74598cb51101e2143097ab270726a561f81f8fd/src/tools.py#L197):
```
 kept = torch.ones(x.shape[0], device=x.device, dtype=torch.bool)
 kept[:-1] = (ranks[1:] != ranks[:-1])
```
while in [QuickCumsumCuda](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/mmdet3d/ops/bev_pool/bev_pool.py#L41):
```
kept = torch.ones(x.shape[0], device=x.device, dtype=torch.bool)
kept[1:] = ranks[1:] != ranks[:-1]
```
**Using `kept[1:]` Seems that can not identify the points which are in boundary throught**
For example:
[image]

Could you explain it in detail?Thanks

kentang-mit, 2022-10-30:
Hi @Z-chocking,

Sorry for the delayed response. I took a couple of days off in the end of last week. The `kept` vector in our implementation is coupled with our CUDA implementation. You will need to also have a look at the `interval_start` and `interval_length` variables in the CUDA kernel:

https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/mmdet3d/ops/bev_pool/src/bev_pool_cuda.cu#L30

In your specific example, for point 1, the `interval_start` will be 0 and `interval_length` will be 2. This will lead to a correct behavior of summing up features 0 & 1 and write back to grid 0. Similar for all other points. I remember that we checked the correctness of our implementation against LSS probably eight or nine months ago, at that time the numerical difference is almost zero.

Best,
Haotian

Z-chocking, 2022-11-17:
Thanks for your patient explanation,i will take a closer look later.Have a nice day.

closed 2022-11-17 by Z-chocking