mit-han-lab/bevfusion #205: Evaluating a model with custom data

VeeranjaneyuluToka, 2022-10-27:
@kentang-mit , I am currently looking into the evaluation/test/visualize, looks like they are taking input as GT tensors along lidar point cloud and images (basically validation samples from NuScenes dataset). I suppose we should be able test with only LiDAR points and camera images, is not it?. I am planning to test the models with my own images and lidar point could, it would be great help if you can provide some suggestions for the same. Thanks!

kentang-mit, 2022-10-30:
Hi @VeeranjaneyuluToka,

Sorry for the delayed response, I took a couple of days off in the end of last week. Your understanding is correct, our current implementation does loads LiDAR points and GT labels even if it is unnecessary. For the GT labels, it is possible not to pass them to the models and the entire inference pipeline will still work: you can try removing lines related to `gt_*` in the configuration file:

https://github.com/mit-han-lab/bevfusion/blob/0e5b9edbc135bf297f6e3323249f7165b232c925/configs/nuscenes/default.yaml#L174

And adjust your dataset file accordingly (for example, lines 277-278 in this file https://github.com/mit-han-lab/bevfusion/blob/0e5b9edbc135bf297f6e3323249f7165b232c925/mmdet3d/datasets/nuscenes_dataset.py#L277 will be unnecessary if you do not need GT).

Best,
Haotian

kentang-mit, 2022-11-03:
Hi @VeeranjaneyuluToka, 

Your understanding is partially correct. Some of the fields are unnecessary in some cases (for example, if you are doing camera-only detection, LiDAR points are definitely unnecessary), in our implementation we keep all the fields just to reduce redundant code (otherwise you will have a camera-only class, a LiDAR-only class and a fusion class, which is not elegant). But other than images and points, you will also have to provide camera intrinsic and extrinsic matrices, augmentation matrices (image augmentation matrices also record the information about how you scale the camera inputs, so they are necessary) during evaluation. I would recommend you to have a look at our forward function: https://github.com/mit-han-lab/bevfusion/blob/0e5b9edbc135bf297f6e3323249f7165b232c925/mmdet3d/models/fusion_models/bevfusion.py#L207. 

All the fields specified in this function (other than the fields not relevant to the modality you are going to use) will be necessary. 

Best,
Haotian

kentang-mit, 2022-11-17:
Closed due to inactivity.

closed 2022-11-17 by kentang-mit