mit-han-lab/bevfusion #207: reproduced the performances is lower

MakiNaruto, 2022-10-30:
Thanks for releasing the awesome codes! I'm a new rookie to be honest and I am trying to reproduce the performances. I have tried training convfuser.yaml use command
 ```torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml ```，

but get mAP is 62.7， Far from 68.5，is that i missed something or need to be load some pretrained model？look forward your answer，thanks。

kentang-mit, 2022-10-30:
Hi @MakiNaruto,

Yes you missed something. You need to load the pretrained LiDAR-only checkpoint. And currently we haven't released the schedule we used for fusion models, but you can infer it from available information.

Best,
Haotian

MakiNaruto, 2022-10-31:
> You need to load the pretrained LiDAR-only checkpoint.

sorry，i still confused，is that use command or edit yaml to add checkpoint？i tried both but fails，I would very appreciate if u can tell me some operation details，thanks！

kentang-mit, 2022-10-31:
Just use `--load_from` in the command line will be a good idea.

closed 2022-11-02 by MakiNaruto