mit-han-lab/bevfusion #208: Transfusion-L

minrui-hust, 2022-10-30:
In transfusion.py: create_2D_grid
this function create 2d grid of shape (2,W,H) then reshape to (1,2,W*H), it is inconsistent with bev_feat of (B, C, H, W) flatten to be (B, C, H*W), which H comes first. After flatten, x and y cord is swapped. 
I read the source code of Transfuison, it is the same.
is this a bug or just on purpose?

kentang-mit, 2022-10-30:
Hi @minrui-hust,

We've checked that the current implementation in our codebase is correct. We did change the order of H and W in the head on purpose (at least for CenterHead). Specifically, in the original implementation, the voxel features are assumed to be in B,C,Z,Y,X; but in our implementation we assume it to be in B,C,X,Y,Z.

Best,
Haotian

minrui-hust, 2022-10-31:
Got it, thanks for your reply,  It confused me for a while.

closed 2022-10-31 by minrui-hust