mit-han-lab/bevfusion #209: Reproduction problems on map segmentation

jiajiaen, 2022-10-31:
Thanks for the great work. 
I had the same problem as [https://github.com/mit-han-lab/bevfusion/issues/196]when reproducing bevfusion-map-segmentation. I loaded the swint-unimages-pretrained.pth and lidar-only-seg.pth weights to initialize the model, and added the following configuration information to fusion-bev256d2-lss.yaml:

      optimizer_config:
          grad_clip:
                   max_norm: 35
                   norm_type: 2

     lr_config:
           policy: cyclic

     momentum_config:
           policy: cyclic
I run the following script:  torchpack dist-run -np 8 python tools/train.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml
but as in [https://github.com/mit-han-lab/bevfusion/issues/196], my training results are not ideal, I suspect this was due to the wrong config file I added.

My following work is based on the bevfusion. I'm really in great need of the config file. Could you please send me a config file.

best,
Jiapeng

kentang-mit, 2022-10-31:
Please send us an Email about that. 

Best,
Haotian

closed 2022-10-31 by kentang-mit

jiajiaen, 2022-10-31:
Thank you very much for your prompt reply, my Email is:  ***@***.***

Best,
Jiapeng

---- Replied Message ----
| From | Haotian (Ken) ***@***.***> |
| Date | 10/31/2022 23:08 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [mit-han-lab/bevfusion] Reproduction problems on map segmentation (Issue #209) |

Please send us an Email about that.

Best,
Haotian