mit-han-lab/bevfusion #210: About  GlobalRotScaleTrans

MaRongbo, 2022-10-31:
Dear authors, thanks for your great work!
I have 2 questions:
1. Why points `rotate(-theta)` but gt_boxes `rotate(theta)` ？
2. In `transform[:3, :3] = rotation.T * scale`, why use rotation.T？ I think for a rotation matrix, its transposition is actually its inverse。
3. [image]

kentang-mit, 2022-10-31:
Hi @MaRongbo,

This is related to whether the transformation matrix is left-multiplied or right-multiplied to the coordinates. What I remember is that the box transformation matrices are right-multiplied, and all others (e.g. points, `lidar_aug_matrix` used by us are left-multiplied. The left-multiplied rotation matrices are exactly the transposed right-multiplied rotation matrices, and it corresponds to a negation in the rotation (it's not hard to derive that from the sin, cos relationships). This is the reason why you see lines 141, 146 and 151. Hope that explanation will help.

Best,
Haotian

MaRongbo, 2022-11-10:
I see, many thanks for your replay~

kentang-mit, 2022-11-10:
No problem. I'll close this issue for now since it is resolved. Feel free to reopen if you have further questions.

closed 2022-11-10 by kentang-mit

HuangVictorAuto, 2023-05-06:
@MaRongbo , have you figured out why they have different theat in the code?

assume we have rotated the pointcloud as well as the gt box clockwise **theta** degree.

1. then points are correctly rotated counter clockwise  **-theta**.
the defualt counter clockwise rotation matrix is [[cos, -sin], [sin, cos]] which is the same as the code in the base_points.py rotate definition
` 
data["points"].rotate(-theta)

elif axis == 2 or axis == -1:
                rot_mat_T = rotation.new_tensor(
                    [[rot_cos, -rot_sin, 0], [rot_sin, rot_cos, 0], [0, 0, 1]]
                )
self.tensor[:, :3] = self.tensor[:, :3] @ rot_mat_T
`
2. the box should rotate clockwise **theta**, but following code show rotate counter clockwise **theta**.
`
rotation = rotation @ gt_boxes.rotate(theta).numpy()

if angle.numel() == 1:
            rot_sin = torch.sin(angle)
            rot_cos = torch.cos(angle)
            rot_mat_T = self.tensor.new_tensor(
                [[rot_cos, -rot_sin, 0], [rot_sin, rot_cos, 0], [0, 0, 1]]
            )
self.tensor[:, :3] = self.tensor[:, :3] @ rot_mat_T
`