mit-han-lab/bevfusion #211: Visualization

zeyuwang615, 2022-11-02:
Hi Kenatang, thanks for the great work!

Could you please provide an example visualization script (and possibly a short clarification on the meaning of the arguments in _tools/visualize.py_)? Thanks a lot!

kentang-mit, 2022-11-03:
```bash
torchpack dist-run -np 1 python tools/visualize.py [config file] --checkpoint [checkpoint file] --out-dir [output dir] --mode [pred/gt] --bbox-score 0.1
```

You can try out the command above and see whether it works. All arguments other than `bbox-score` are quite self-explanatory, for the `bbox-score`, it is a threshold filtering out low-confidence predictions. You can select this value according to the visual quality of your generated figures. A higher threshold will remove more duplicate predictions and create a cleaner visualization.

kkangshen, 2022-11-07:
hi @kentang-mit   thanks for the great work!

When I visualize the segmentation results, the following error occurs when I run the following command, I want to know what is the reason for this

`python visualize.py ../configs/nuscenes/seg/fusion-bev256d2-lss.yaml --checkpoint=pretrained/bevfusion-seg.pth
`
```
Traceback (most recent call last):
  File "visualize.py", line 185, in <module>
    main()
  File "visualize.py", line 101, in main
    outputs = model(**data)
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 50, in forward
    return super().forward(*inputs, **kwargs)
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/torch/nn/parallel/data_parallel.py", line 166, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/mnt/data/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/mnt/data/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 231, in forward_single
    feature = self.extract_camera_features(
  File "/mnt/data/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 114, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/mnt/data/bevfusion/bevfusion-main/mmdet3d/models/vtransforms/lss.py", line 76, in forward
    x = super().forward(*args, **kwargs)
  File "/home/ubuntu/miniconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func
    return old_func(*args, **kwargs)
TypeError: forward() takes 11 positional arguments but 12 were given
```

kentang-mit, 2022-11-07:
Hi @kkangshen,

Let's try inserting 
```python
metas,
```
after this line of code: https://github.com/mit-han-lab/bevfusion/blob/da751a2c9fa7d436f4ba22427ded5782f3b9a50e/mmdet3d/models/vtransforms/base.py#L177. If that works for you I will publish a PR to fix the bug.

Best,
Haotian

Wolfwjs, 2022-11-08:
Hi  @kentang-mit ,

Could you please share how to make such demo? Is there any script for reference? Thank you!

[image]

kentang-mit, 2022-11-09:
@IAMShashankk, glad to hear that.

@Wolfwjs, that demo is not generated in a fully automated manner. The idea is to first visualize predictions on each frame of the validation set (which our `tools/visualize.py` can help you with), we then use ffmpeg to create a video demo. After that we make some manual edits to that video, adding in the background and legends.

kentang-mit, 2022-11-17:
Closed due to inactivity.

closed 2022-11-17 by kentang-mit