mit-han-lab/bevfusion #215: Incorrect eval result

AlmoonYsl, 2022-11-03:
I get a wrong results when I evaluate the pretrained checkpoints downloaded from the provided url. I'm confused what I did wrong.
Btw, I use the ```test.py``` to evaluate the model instead of directly reading out the results from the result file.
mAP: 0.6583
mATE: 0.2897
mASE: 0.2554
mAOE: 0.3089
mAVE: 0.2557
mAAE: 0.1901
NDS: 0.6992

happinesslz, 2022-11-03:
Same problem with you.  could you provide any suggestions? @zhijian-liu @tonylins   @synxlin

kentang-mit, 2022-11-03:
When did you clone the code? As I have mentioned in previous issues, please make sure that you clone the latest version and use the latest pretrained checkpoints. Old codebase + new checkpoint / new codebase + old checkpoint will not work.

AlmoonYsl, 2022-11-03:
> When did you clone the code? As I have mentioned in previous issues, please make sure that you clone the latest version and use the latest pretrained checkpoints. Old codebase + new checkpoint / new codebase + old checkpoint will not work.

Thanks for your replay! I clone the main branch of the repo and download the checkpoints by this url (  ```https://bevfusion.mit.edu/files/pretrained_updated/bevfusion-det.pth```) today. So, I think all things are latest.

kentang-mit, 2022-11-03:
Did you also have a chance to check whether all the library versions match with our description? There is one issue before specifically mentioned that if you did not use `Pillow==8.4.0`, there might be problems related to the reproducibility. You can try out our Docker and see whether the same problem still exists.

AlmoonYsl, 2022-11-03:
> Did you also have a chance to check whether all the library versions match with our description? There is one issue before specifically mentioned that if you did not use `Pillow==8.4.0`, there might [...]

I have kept all the library versions match with the provided Dockerfile except cuda. My cuda version is 11.1 and I'm not sure if this will have an impact on performance. I will try out Docker to check the result again. Thanks!

AlmoonYsl, 2022-11-04:
> Did you also have a chance to check whether all the library versions match with our description? There is one issue before specifically mentioned that if you did not use `Pillow==8.4.0`, there might [...]

I still got the worse results (mAP: 0.6583, NDS: 0.6991) after using the provided Docker. The only difference is that I use two NVIDIA RTX 3090 gpus to evaluate the model.

AlmoonYsl, 2022-11-04:
> Did you also have a chance to check whether all the library versions match with our description? There is one issue before specifically mentioned that if you did not use `Pillow==8.4.0`, there might [...]

I get the correct results (mAP: 68.55, NDS: 71.44) by changing the downloaded checkpoints. I think the provided url is incorrect. Thanks!

closed 2022-11-04 by AlmoonYsl