mit-han-lab/bevfusion #218: Meaning of the 'img_aug_matrix'and 'lidar_aug_matrix'

zhenghw111, 2022-11-04:
Hi, thanks for opening access the nice work!
What is the meaning of the 'img_aug_matrix' and 'lidar_aug_matrix' ?

kentang-mit, 2022-11-04:
Hi @zhenghw111,

`img_aug_matrix` means the transformation matrix applied to the camera inputs including rotation, flipping and translation. Each can be represented as a 3x3 matrix (changing x, y coordinates of the pixels and keep the z coordinates unchanged).  Similarly `lidar_aug_matrix` records the transformations applied to the point cloud input. 

Best,
Haotian

zhenghw111, 2022-11-04:
Thanks really! Does `img_aug_matrix` mean the camera extrinsics matrix of the camera?

kentang-mit, 2022-11-04:
`camera2lidar` is the extrinsics you are looking for. `img_aug_matrix` happens in the pixel coordinate system.

kentang-mit, 2022-11-17:
Closed due to inactivity.

closed 2022-11-17 by kentang-mit