mit-han-lab/bevfusion #219: Error in nuscenes test data

study0101, 2022-11-04:
I can use test.py to evaluation on val data, but when I use nuscenes test data I get error.

dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox

The program error is as follows：

  File "tools/test.py", line 231, in <module>
    main()
  File "tools/test.py", line 204, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/usr/local/lib/python3.8/dist-packages/mmdet/apis/test.py", line 96, in multi_gpu_test
    for i, data in enumerate(data_loader):
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/usr/local/lib/python3.8/dist-packages/torch/_utils.py", line 434, in reraise
    raise exception
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/usr/local/lib/python3.8/dist-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/dataset/bevfusion/mmdet3d/datasets/custom_3d.py", line 291, in __getitem__
    return self.prepare_test_data(idx)
  File "/dataset/bevfusion/mmdet3d/datasets/custom_3d.py", line 180, in prepare_test_data
    input_dict = self.get_data_info(index)
  File "/dataset/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 277, in get_data_info
    annos = self.get_ann_info(index)
  File "/dataset/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 300, in get_ann_info
    mask = info["num_lidar_pts"] > 0
KeyError: 'num_lidar_pts'

--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[49443,1],0]
  Exit code:    1
--------------------------------------------------------------------------

kentang-mit, 2022-11-04:
Hi @study0101,

If you want to run inference on the test set, you will have to remove anything related to GT loading (including this field, `num_lidar_pts`) in the data preprocessing pipeline. This is because these fields are not released by nuScenes.

Best,
Haotian

closed 2022-11-04 by kentang-mit

youngfly, 2023-03-20:
Dear author, we are curious about "remove anything related to GT loading (including this field, num_lidar_pts) in the data preprocessing pipeline".
Please show us how to deal with it?
Best wishes