mit-han-lab/bevfusion #220: Info for testing BEVfusion on nuscenes v1.0-mini

MicheleRep, 2022-11-04:
Hi,
I'm trying to explore the BEVfusion library. I set up the docker container and within it I prepared the nuscenes dataset v1.0-mini specifying it version with the following command:

`python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes --version v1.0-mini`

But when I execute:

`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`

or

`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --show`

I obtain the following feedback and it seems stuck without giving any output.

[image]

My questions are: should it print or output something? Should I wait for it or is clearly stuck?

My system is an MSI Creator M16 A12UD, ram 16gb, CPU Intel Core i7-12700H 12th Gen, GPU Nvidia GeForce RTX 3050 Ti Mobile 4gb driver version: 520.61.05.

Thanks a lot.

kentang-mit, 2022-11-04:
Hi @MicheleRep,

I've never tried `v1.0-mini`, but when you see the process stuck, I think you can add the `-v` flag to your command and try again (`torchpack -np ... -v python ...`). Usually this will give you some clearer error. By the way, the GPU memory of 3050Ti is too small, 4GB is not enough for running BEVFusion inference. In the future we will release a more memory-efficient version of BEVFusion which will allow you to run inference under 4GB of GPU memory, but I'm a bit packed recently and do not have the time to start preparing the code immediately.

Best,
Haotian

MicheleRep, 2022-11-04:
Thank you Haotian, for the answer and for all the work you do.

Sincerely,
Michele.

kentang-mit, 2022-11-04:
No problem and I'll close this issue and mark it as completed.

closed 2022-11-04 by kentang-mit