mit-han-lab/bevfusion #221: Unbalanced memory usage in multi-card training

sunnyHelen, 2022-11-04:
When training in multiple GPU cards, it seems that the memory usage is always unbalanced. The highest usage of the 0th card causes my batch size can only reach 2. Do you know how to solve this problem?
[image]

kentang-mit, 2022-11-04:
Hi @sunnyHelen,

I didn't observe such unbalanced memory consumption. I remember that 40G is just enough to run batch=4 on our machine for the current version of this codebase. Maybe you can try turning off `cudnn.benchmark` and see if the memory consumption will go down.

Best,
Haotian

kentang-mit, 2022-11-17:
Closed due to inactivity.

closed 2022-11-17 by kentang-mit