mit-han-lab/bevfusion #224: Error during evaluation on nuScenes

YoushaaMurhij, 2022-11-10:
Thanks for publishing your work!
I ran this command inside the container that your provided:
```bash 
torchpack dist-run -np 2 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
```
but got this error:
```bash
File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/opt/conda/lib/python3.8/site-packages/mmdet/apis/test.py", line 98, in multi_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/trainer/bevnet/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/trainer/bevnet/mmdet3d/models/fusion_models/bevfusion.py", line 230, in forward_single
    feature = self.extract_camera_features(
  File "/home/trainer/bevnet/mmdet3d/models/fusion_models/bevfusion.py", line 114, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/trainer/bevnet/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/trainer/bevnet/mmdet3d/models/vtransforms/base.py", line 250, in forward
    cur_coords = torch.inverse(cur_lidar_aug_matrix[:3, :3]).matmul(
RuntimeError: cusolver error: CUSOLVER_STATUS_EXECUTION_FAILED, when calling `cusolverDnSgetrf( handle, m, n, dA, ldda, static_cast<float*>(dataPtr.get()), ipiv, info)`
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[56338,1],0]
  Exit code:    1

```
Any suggestion? what could lead to such an error?
Thank.

kentang-mit, 2022-11-10:
Probably you can have a look at this issue: https://github.com/pytorch/pytorch/issues/70122.

YoushaaMurhij, 2022-11-10:
Thanks. I will check this issue.

YoushaaMurhij, 2022-11-11:
Yes, downgrading cuda to 11.1 solved the problem for me. Upgrading the server cuda drive version will a better solution.
Thanks for your help!.
I just wanted to add that you don't need to use miniconda inside the docker container.
```dockerfile
RUN apt install -y python3-pip
RUN pip3 install torch==1.10.1+cu113 torchvision==0.11.2+cu113 \
          -f https://download.pytorch.org/whl/cu113/torch_stable.html
```

closed 2022-11-11 by YoushaaMurhij

hankeceli, 2023-12-19:
Hi @kentang-mit, 
I've encountered an issue similar to the one discussed in this flood. The error pertains to a line of code involving matrix multiplication (matmul): `cur_coords = torch.inverse(cur_lidar_aug_matrix[:3, :3]).matmul(
RuntimeError: cusolver error: CUSOLVER_STATUS_INTERNAL_ERROR, when calling cusolverDnCreate(handle)`

Your suggestion in a similar issue from a year ago recommended downgrading the CUDA version. My current CUDA version is 12.3 and torch version is 11.1

Given that the advice is from a year ago from you, is it still reasonable to consider downgrading the CUDA version as a potential solution? Any insights or alternative recommendations would be greatly appreciated.

Thank you/