mit-han-lab/bevfusion #226: Reproduction of the LiDAR branch

Zhongdao, 2022-11-12:
Hi all,

About two months ago I opened an issue (#99) and asked if anyone reproduced the reported results of the LiDAR branch. 
Back to then, the training details were not released and my results were around 2 point lower.
Now that the training details of the LiDAR branch are all released, I tried the new code and found the results still not reproducible (62.5% v.s. 64.7% mAP on nuScenes val).

I'm pretty sure that I followed the instructions and had not modified the code. 
Curious to know if anyone reproduced the result, could you share some information?

kentang-mit, 2022-11-13:
Hi @Zhongdao,

There are several things you can try. First, please run the test script separately after training the model, the validation accuracy reported during training is lower than the correct value and I have not figured out the reason. Second, I did notice that in some runs the validation accuracy could drop to the 63.x range. You probably need to train the model more than once   and see if you can get a result close to what is reported in the pretrained model (64.7 mAP). 

By the way, the training schedule of the LiDAR branch is almost the same as the one proposed by the authors of TransFusion. The only difference is that instead of manually restart at epoch 15 and switch to the "fade strategy" with a smaller starting lr and a full schedule, we automatically enables the "fade strategy" and continues the original lr schedule. This might be influential, so what you can do is that you can save the model at epoch 15 and also try out the strategy from the official TransFusion repo. 

At the end of the day, this is a problem related to TransFusion reproduction, not that much to our method. Hope that my explanations will be helpful to you.

Best,
Haotian

Zhongdao, 2022-11-14:
@kentang-mit 
Thanks a lot for answering! 
I understand the LiDAR branch is the same as in TransFusion, and I definitely believe the results are repreducable.
Actually I'm not blaming you guys for the code not being reproducable, I'm just curious if someone make it to reproduce the results, so I would know if the problem is on my side (could be the data or something else).  I like this code base and want to develop something upon it, and that's why I care about the LiDAR branch so much.
Will try the two points you mentioned! 
By the way, do you still have the logs of the LiDAR branch training? It would be helpful if I can get access to it.

kentang-mit, 2022-11-14:
Hi @Zhongdao,

[output.txt](https://github.com/mit-han-lab/bevfusion/files/10004338/output.txt)

The txt file attached above is the training log we had around seven months ago. As you can see, there is a quick accuracy ramp up after the fading strategy was turned on, and this might be the reason why the results are not so stably reproducible. I would definitely recommend you to save the predictions at epoch 15 and try more times to see whether it works. On other datasets where we do not use the fading strategy, I usually observe that the results for LiDAR-only are more stable.

Best,
Haotian

wzc9401, 2022-11-24:
@Zhongdao Hi, have you reproduce the lidar-only performance? I meet the same problem.

Zhongdao, 2022-11-30:
@wzc9401 
Hi.
Still in progress. Currently I am guessing this is somehow related to the CUDA version. What is your CUDA version?
I am using CUDA 10.2 (I have to use this version due to some limit) and cannot reproduce. 
But I know someone made it to reproduce using cuda 11.2.

kentang-mit, 2022-11-30:
Hi @Zhongdao,

We only tested the codebase with CUDA 11.3. One possible reason for discrepancy in results is that the TF32 tensor core matrix multiplication is not supported until CUDA 11.0, and it is turned on by default for FP32 training in PyTorch. This means that if use use CUDA 10.x, the training behavior is guaranteed to be different from others. But usually this will not lead to large discrepancy in accuracy. I believe the fundamental reason is that the GT paste fading strategy brings about some instability.

Best,
Haotian

Zhongdao, 2022-11-30:
Thanks Haotian!
Will do some experiments in the next few weeks and keep the progress posted in this issue!

wzc9401, 2022-12-01:
> @wzc9401 Hi. Still in progress. Currently I am guessing this is somehow related to the CUDA version. What is your CUDA version? I am using CUDA 10.2 (I have to use this version due to some limit) and cannot reproduce. But I know someone made it to reproduce using cuda 11.2.

@Zhongdao @kentang-mit 
Hi, my CUDA version is 10.2, I will try cuda 11.3. thanks!

huzhangcs, 2022-12-26:
> Thanks Haotian! Will do some experiments in the next few weeks and keep the progress posted in this issue!

Did you get some recent progress on this issue？I met the same problem on lidar branch: 62.6 vs 64.7，2 points lower than reported. My cuda version is 11.1. Thank you .

Zhongdao, 2023-01-03:
@huzhangcs 
I tried syncBN. It brought some improvements, and now my best result is around 63.5 mAP

kentang-mit, 2023-01-05:
I see. Actually we did not use SyncBN in our experiments. Let me try to find out a LiDAR-only schedule that can reproduce the 64.7 results more easily.

closed 2023-03-30 by kentang-mit