mit-han-lab/bevfusion #227: About fusion branch

SlierYang, 2022-11-14:
Thanks for your awesome job! 
When I tried to reproduce the fusion results on the validation dataset, I ran into a problem. Based on lidar pre-training and nuimage pre-training, I fine-tuned 6 epochs, but only got NDS 0.70 and mAP 0.66. When fine-tuning, should I freeze part of the weights like transfusion? Or maybe I missed something?

kentang-mit, 2022-11-14:
No, you do not need to freeze anything. I'll release the schedule soon, basically it's cosine annealing lr with all augmentations on. Definitely remember to evaluate again using `test.py` instead of directly read out what is shown at the end of training.

SlierYang, 2022-11-16:
T

SlierYang, 2022-11-16:
Thanks for your reply! I have tried another training with CosineAnnealing(AdamW lr 1.0e-4, 6 epochs, without grad clipping), and got a higher performance(NDS 0.71). By the way, how do you combine test time augmentation with other pipelines? I have tried to add MultiScaleFlipAug3D to test_pipeline, but failed. Can you give me some advice to use tta? Thanks a lot!

kentang-mit, 2022-12-12:
Closed due to inactivity.

closed 2022-12-12 by kentang-mit

liyih, 2023-12-04:
@SlierYang I am wondering have you add MultiScaleFlipAug3D to Bevfusion, and how do you achieve that?