mit-han-lab/bevfusion #228: RuntimeError: sigmoid_focal_loss_forward_impl: implementation for device cuda:0 not found.

YoushaaMurhij, 2022-11-14:
Hi, I tried to train the model for LiDAR-only detector using this command:
```bash
torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml
```
but got the following error. All the other training commands are working fine exept this one.
Do I need to build any additional library? Any suggestion? 
Thanks.

```bash
Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 76, in main
    train_model(
  File "/home/trainer/bevnet/mmdet3d/apis/train.py", line 126, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/usr/local/lib/python3.8/dist-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/trainer/bevnet/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/usr/local/lib/python3.8/dist-packages/mmcv/parallel/distributed.py", line 52, in train_step
    output = self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/trainer/bevnet/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/trainer/bevnet/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/usr/local/lib/python3.8/dist-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/trainer/bevnet/mmdet3d/models/fusion_models/bevfusion.py", line 269, in forward_single
    losses = head.loss(gt_bboxes_3d, gt_labels_3d, pred_dict)
  File "/usr/local/lib/python3.8/dist-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/trainer/bevnet/mmdet3d/models/heads/bbox/transfusion.py", line 645, in loss
    layer_loss_cls = self.loss_cls(
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/mmdet/models/losses/focal_loss.py", line 233, in forward
    loss_cls = self.loss_weight * calculate_loss_func(
  File "/usr/local/lib/python3.8/dist-packages/mmdet/models/losses/focal_loss.py", line 139, in sigmoid_focal_loss
    loss = _sigmoid_focal_loss(pred.contiguous(), target.contiguous(), gamma,
  File "/usr/local/lib/python3.8/dist-packages/mmcv/ops/focal_loss.py", line 55, in forward
    ext_module.sigmoid_focal_loss_forward(
RuntimeError: sigmoid_focal_loss_forward_impl: implementation for device cuda:0 not found.

```

kentang-mit, 2022-11-14:
Hi @YoushaaMurhij,

Based on my experience, this error is related to mmcv and mmdetection versions that might be inconsistent with this codebase. Please double check whether you have mmcv==1.4.0, mmcv-full==1.4.0 and mmdet==2.20.0 installed on your machine. Please also make sure that you ran 
```python
python setup.py develop
```
under the root directory of our codebase before starting any experiments. 

Best,
Haotian

YoushaaMurhij, 2022-11-14:
@kentang-mit , thanks for your reply. I am using those versions (actually using your Dockerfile). I ran that command in the root. And all the other train/test commands are working fine except for this one. I tried another version of mmcv but with no luck.

kentang-mit, 2022-11-14:
That is a bit strange. May I have some knowledge about your GPU type, CUDA library version, etc.?

YoushaaMurhij, 2022-11-15:
Sure, Tesla V100 * 8, Driver Version: 450.119.03   CUDA Version: 11.0 (Host versions). and inside the container, the same as yours.

kentang-mit, 2022-11-15:
I see, would you mind trying out CUDA version 11.3 on the host side? Currently I don't have access to GPUs with Volta architectures, we've only tested with Turing and Ampere.

YoushaaMurhij, 2022-11-16:
I will try that.

mpQuintana, 2022-11-22:
Hello @YoushaaMurhij, did you manage to solve it? 
I get the same error. I am also using the docker with host Ubuntu20.04, Cuda 11.3, RTX3090, Driver Version: 510.85.02.

YoushaaMurhij, 2022-11-22:
I haven't tried yet. I will check it during the next week.

djiajunustc, 2022-12-09:
@YoushaaMurhij I use the same GPU and the same driver/Cuda version as yours and have the same problem. Have you solved it?

YoushaaMurhij, 2022-12-09:
Sorry, I did not have enough time to work on it. Could you please try Haotian's suggestions. I will write as soon as I fix this problem. But, I can not guarantee that in this month.

kentang-mit, 2022-12-10:
Please let me know if the solution will not work. We are doing a major reformat on this codebase internally to remove unnecessary dependencies on mmcv/mmdet and make the installation process easier, but this reformat process might take relatively long time (months).

kentang-mit, 2022-12-27:
Closed due to inactivity.

closed 2022-12-27 by kentang-mit

YoushaaMurhij, 2022-12-30:
@kentang-mit, @djiajunustc, @mpQuintana, upgrading to cuda 11.8 solved the problem for me!
`Driver Version: 520.61.05    CUDA Version: 11.8`.

ztf-ucas, 2023-01-31:
> Ampere

Hi～, have you solved it?

ztf-ucas, 2023-01-31:
> 

@djiajunustc Hi~ have you solved it ?

ssuralcmu, 2023-03-26:
> @kentang-mit, @djiajunustc, @mpQuintana, upgrading to cuda 11.8 solved the problem for me! `Driver Version: 520.61.05 CUDA Version: 11.8`.

Hi, did you upgrade CUDA within the docker or for the overall PC?
1. In case it was within the docker, did you have to change the driver version in the PC? Also, what changes were needed in the Dockerfile?
2. If the changes were made for the overall PC, what changes do we need in the Dockerfile?

YoushaaMurhij, 2023-03-27:
> 

This is my base image for Dockerfile  `nvidia/cuda:11.3.1-cudnn8-devel-ubuntu20.04`
and I am using Driver `Version: 520.61.05 CUDA Version: 11.8`  on the host.

HenryZhangJianhe, 2023-06-26:
> > 
> 
> This is my base image for Dockerfile `nvidia/cuda:11.3.1-cudnn8-devel-ubuntu20.04` and I am using Driver `Version: 520.61.05 CUDA Version: 11.8` on the host.

could you share your docker image, I upgrade v100 host pc cuda version as yours, but it still arise this error.

BigGold0202, 2023-07-04:
> > > 
> > 
> > 
> > This is my base image for Dockerfile `nvidia/cuda:11.3.1-cudnn8-devel-ubuntu20.04` and I am using Driver `Version: 520.61.05 CUDA Version: 11.8` on the host.
> 
> could you share your docker image, I upgrade v100 host pc cuda version as yours, but it still arise this error.

I solve this by reinstall mmcv and mmcv-full with `MMCV_WITH_OPS=1 FORCE_CUDA=1`.
The specific script for the docker environment is:
`MMCV_WITH_OPS=1 FORCE_CUDA=1 pip install mmcv==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html`
`MMCV_WITH_OPS=1 FORCE_CUDA=1 pip install mmcv-full==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html`
I'm using A100 with host cuda version 11.4. After reinstall, you can use `mmcv.utils.collect_env()` to check. The `mmcv cuda compiler` should be 11.3.

YongYongLi123, 2024-01-29:
> I solve this by reinstall mmcv and mmcv-full with `MMCV_WITH_OPS=1 FORCE_CUDA=1`. The specific script for the docker environment is: `MMCV_WITH_OPS=1 FORCE_CUDA=1 pip install mmcv==1.4.0 -f [...]
I adopted this method to solve the problem

YongYongLi123, 2024-01-29:
> I solve this by reinstall mmcv and mmcv-full with `MMCV_WITH_OPS=1 FORCE_CUDA=1`. The specific script for the docker environment is: `MMCV_WITH_OPS=1 FORCE_CUDA=1 pip install mmcv==1.4.0 -f [...]

Hello, I have used this method to solve the problem

hankeceli, 2024-03-02:
> I solve this by reinstall mmcv and mmcv-full with `MMCV_WITH_OPS=1 FORCE_CUDA=1`. The specific script for the docker environment is: `MMCV_WITH_OPS=1 FORCE_CUDA=1 pip install mmcv==1.4.0 -f [...]

It seems mmcv 1.4.0 wheels don't exist anymore in https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html
Any suggestions on how to download them again?