mit-han-lab/bevfusion #229: Question about TransFusionHead for camera-only detector.

JoeyforJoy, 2022-11-14:
Have you tried `TransFusionHead` for Camera-Only detector? I noticed that LiDAR-only and fusion models all utilize it. Why choose `CenterHead` specially for Camera-Only model? Related ablation study was not found in your paper.

kentang-mit, 2022-11-14:
Yes, I have tried. TransFusion head for camera-only does not give as good performance as CenterHead. We did not report the results because we did not have enough time to tune the parameters for it, thus it would be unfair to report a relatively low results for TransFusion.

The observation in general is that TransFusion head requires relatively good performance with the CenterHead (say, LiDAR-level mAP like 55-65). This is because TransFusion head is basically similar to two-stage detectors, where if you do not have an accurate first stage the results can be bad. Hope that explanation will be helpful.

JoeyforJoy, 2022-11-14:
您好，您的邮件已经收到，我会尽快给您回复。

JoeyforJoy, 2022-11-17:
Thanks for your quick reply. Well, you may be right. The init positions of object queries in TransFusion head are highly relied on the BEV feature. The camera-only model may struggle in the lack of depth informantion, which may lead to bad initialization.

JoeyforJoy, 2022-11-17:
I found similar experiments in a similar research (BEVFusion: A Simple and Robust LiDAR-Camera Fusion Framework). They concluded same results.
[image]

kentang-mit, 2022-11-17:
Hi @JoeyforJoy,

I think their conclusion is similar to ours, but the absolute values of mAP and NDS might be a bit low for the TransFusion head. I suspect that in order to achieve a reasonable performance for the TransFusion head, you probably need to first pretrain their heatmap regressor, and then jointly finetune it with the box dimension / position regressors.

Best,
Haotian

closed 2022-11-28 by JoeyforJoy