mit-han-lab/bevfusion #230: How to evaluate objects with respect to object distances

anti-destiny, 2022-11-15:
Hi. I noticed that you evaluated your model under different object sizes and distances in Table 3. I want to know how to achieve this with the original nuScenes evaluation API. Do I need to do some modifications to the source code of nuScenes?

kentang-mit, 2022-11-15:
The answer to this question should be similar to https://github.com/mit-han-lab/bevfusion/issues/139. You will have to customize the nuScenes evaluation API, basically doing some filtering before calculating the metrics.

hollow-503, 2022-11-25:
Maybe you refer to function `filter_eval_boxes` in [nuscenes-devkit](https://github.com/jinfagang/nuscenes-devkit/blob/25e430a50c404311cdcb2759190c9d33eee9497d/python-sdk/nuscenes/eval/detection/loaders.py). Hope that will help you!