mit-han-lab/bevfusion #234: onnx translate tensorrt error

FlyEgle, 2022-11-16:
I have finsh the torch2onnx, and the value is same with torch version, like this
torch result
[image]
onnxruntime result
[image]

But when i use onnx2trt, trt version is 8.1.4.5,there have some bug!!!like this
[image]
It seems the nonzero which is onnx op, but tensorrt not support 
[image]

The problem in **vtransforms/base.py**
[image]

So i must change the op which can convert to onnx & trt ,How can i use the `troch.index_select` to replace the x[kept]?

thanks!

kentang-mit, 2022-11-16:
Hi @FlyEgle,

We haven't tried end-to-end TensorRT conversion yet, but for your problem I believe the best solution is what you have mentioned: calculate `geom_feats` as a preprocessing step and directly store it in the onnx. This might cause some small performance degradation, because `geom_feats` is not the same for all the samples (in fact, it is dependent on the camera intrinsics and extrinsics, and these two matrices changed very rarely on nuScenes). Exporting camera encoder, decoder and vtransform separately to onnx (and then trtengine) will also be a feasible choice.

By the way, `geom_feats` is actually used in training.

Best,
Haotian

FlyEgle, 2022-11-16:
Hi @kentang-mit,

1. the `self.frustum() `have generate the image2points, 
[image]
The points is no grad.
2. points do undo-postprocessing, is also no-training
3. geom_feats change the coordinate and do stack batch idx to make [:, 4]
4. geom_feats filter out values and make ranks for pooling.
5. follow geom_feats presum index to get coordinate and get the x.value.

So I think this is only a spare grid , no need training, why used for training, i have a little confuse.

kentang-mit, 2022-11-17:
Hi @FlyEgle,

I think we had different understanding on "used in training". If your point is "`geom_feats` does not require a gradient", then I believe you are correct. For your question related to intrinsics and extrinsics in the real world, you are right, but for nuScenes there are two sets of these parameters (if I remember it correctly), and this is the reason why I remind you of the possible performance degradation if you regard `geom_feats` as a constant.

By the way, the reason why we directly use `cam2lidar` is that the "ego" for camera is different from the "ego" for LiDAR (reported in previous issues). We found that if we use `cam2lidar @ lidar2ego @ ego2global`, the projection will be incorrect and the performance of camera-only models are actually impacted by this mistake. However, such a modification has very small impact on the performance of fusion models.

Best,
Haotian

FlyEgle, 2022-11-18:
I see, thakn for your replay

closed 2022-11-18 by FlyEgle

daxiongpro, 2022-11-30:
> I have finsh the torch2onnx, and the value is same with torch version, like this torch result [image] onnxruntime result [image]
> 
> But when i use onnx2trt, trt version is 8.1.4.5,there have some bug!!!like this [image] It seems the nonzero which is onnx op, but tensorrt not support [image]
> 
> The problem in **vtransforms/base.py** [image]
> 
> So i must change the op which can convert to onnx & trt ,How can i use the `troch.index_select` to replace the x[kept]?
> 
> I also have a problem, geom_feats is not used to training, if i can remove from forward, make this part as a pre-processing with opencv or eigen C++ library. But i don't know how cuda can accelerate speed?
> 
> thanks!

Hi, 
Could you please share your code about how to convert bevfusion to onnx?
Hope you e-mail me: `zhengyi_cai@foxmail.com`
Thank you very much!

zxc-106, 2023-03-08:
> I have finsh the torch2onnx, and the value is same with torch version, like this torch result [image] onnxruntime result [image]
>
> But when i use onnx2trt, trt version is 8.1.4.5,there have some [...]

hi, could you please share the torch2onnx idea, it would be better if you can share the code.
Hope you can email me: zhaoxuchao123@gmail.com
extremely thank you.

GzpTez0514, 2023-03-14:
> d to training, if i can remove from forward, make this part as a pre-processing with opencv or eigen C++ library. But i don't know how cuda can accelerate speed?
> 
> thanks!

Hello, could you please send me the code of onnx? I can pay. My email is gzptez0514@gmail.com, thank you very much

GzpTez0514, 2023-03-14:
> I have finsh the torch2onnx, and the value is same with torch version, like this torch result [image] onnxruntime result [image]
>
> But when i use onnx2trt, trt version is 8.1.4.5,there have some [...]

Hello, could you please send me the code of onnx? I can pay. My email is gzptez0514@gmail.com, thank you very much

YaqinLong, 2023-03-27:
> I have finsh the torch2onnx, and the value is same with torch version, like this torch result [image] onnxruntime result [image]
>
> But when i use onnx2trt, trt version is 8.1.4.5,there have some [...]

Hi, could you send me the code of export.py that can convert torch to onnx?  I faced some problems. My email is yaqinlong427@gmail.com, thank you!

sandeepnmenon, 2023-06-14:
> I have finsh the torch2onnx, and the value is same with torch version, like this torch result [image] onnxruntime result [image]
>
> But when i use onnx2trt, trt version is 8.1.4.5,there have some [...]

Hi @FlyEgle . I am trying to run a few benchmarks on my modifications to the BEVFusion model in tensorrt. It would be really helpful if you could share how you did the export. 
My email is: menonsandu@gmail.com

igo312, 2024-01-04:
@FlyEgle hi,
There should be  some cuda kernel in the network. How did you deal with it when running onnxruntime ?