mit-han-lab/bevfusion #238: Assertion error during testing

JiSuanJiDaWang, 2022-11-20:
Hi, I ran 'torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/default.yaml pretrained/bevfusion-det.pth --eval bbox' and encountered the following error. Can you give me some advice on how to solve it? Thanks.
[image]

kentang-mit, 2022-11-20:
Hi @JiSuanJiDaWang,

The config you used is not correct. You can have a try using `configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml`. 

Best,
Haotian

JiSuanJiDaWang, 2022-11-21:
Thanks! It works now

closed 2022-11-21 by JiSuanJiDaWang