mit-han-lab/bevfusion #239: ValueError: matrix contains invalid numeric entries

lawrence-cj, 2022-11-21:
[image]
Dear Author, I met this problem during the last iteration of the first epoch. Is there any suggestion about it. Byw, I have saw the issue in https://github.com/mit-han-lab/bevfusion/issues/190#issue-1411031759

lawrence-cj, 2022-11-22:
Actually, this error happened when I trained the Lidar branch.

kentang-mit, 2022-11-22:
Are you sure that you are using exactly the same configuration file we released? The loss value seems to be very different from what I got (the loss value is expected to be around 2.7 after 1 epoch, but yours is close to 8). Besides, clearly you are using a different batch size from ours. With our default setting (batch=4), there should be 3862 iterations per epoch, not 2575. I'd recommend you to start from our configurations first and see whether the results are reproducible, otherwise I don't have enough context to help you with the debugging context.

lawrence-cj, 2022-11-23:
Thank you for your reply! I changed the batch size in the configuration from 4 to 16 and GPU numbers from 8 to 3, then I got the error. I have reproduced the result already and I want to change some hyper parameters. Besides, I only have 3GPUs available.

kentang-mit, 2022-11-23:
I guess one thing you could try will be to hold the "total batch size" constant. Originally the total batch size is 32. In your case you could probably try out batch size = 10 on a single GPU. In this case you do not need to adjust the learning rate. In your case, because the total batch size is increased from 32 to 48, I guess you probably need to enlarge the starting learning rate by 1.5x (according to the conclusion in [this paper](https://arxiv.org/abs/1706.02677), but I haven't tried whether it will work for your case).

lawrence-cj, 2022-11-24:
Appreciate for your reply. I will try your advises. Besides, In the above picture, I thought the problem may be caused by the "linear_sum_assignment" function. Is there any relation between this function and the starting learining rate?

kentang-mit, 2022-11-25:
@lawrence-cj, typically when training does not converge properly, the function you mentioned might return incorrect assignment with nans in the result. Starting learning rate is important for the model behavior in the beginning, so you need to tune it if you have different number of GPUs and different batch size.

lawrence-cj, 2022-11-26:
Thanks a lot for your reply. Grateful for you guys nice work. Problems solved perfectly. Issue closed.

Best. Have a nice day.
Lawrence

closed 2022-11-26 by lawrence-cj

liyejlu, 2023-03-22:
hello, I wonder how to change the learning rate?

lawrence-cj, 2023-03-22:
> hello, I wonder how to change the learning rate?
You may multiply a 0.5 or 0.1 to the learning rate