mit-han-lab/bevfusion #240: cannot train model over multiple GPUs in one machine successfully.

from-Pb2Ag, 2022-11-21:
Hi Zhijian and Haotian,
Thanks for your awesome paper and code!
Currently I have some trouble with training a more customized model (to be specify, the LiDAR-only detector.), before it  I have evaluate the pretrained models successfully in a no-distributed method (by setting the `distributed` variable to `False` in `line 162` of `./tools/test.py`, and comment `dist.init()` in `line 115`.).
As a beginner who has no idea how to run these scripts over multiple GPUs, I still comment `dist.init()` in `line 21` of  `./tools/train.py`, for I will receive such error if I not comment it:
```
Traceback (most recent call last):
  File "/home/yhqian/archives/bevfusion/tools/train.py", line 91, in <module>
    main()
  File "/home/yhqian/archives/bevfusion/tools/train.py", line 24, in main
    dist.init()
  File "/home/yhqian/anaconda3/envs/BEVFusionEnv/lib/python3.8/site-packages/torchpack/distributed/context.py", line 23, in init
    master_host = 'tcp://' + os.environ['MASTER_HOST']
  File "/home/yhqian/anaconda3/envs/BEVFusionEnv/lib/python3.8/os.py", line 673, in __getitem__
    raise KeyError(key) from None
KeyError: 'MASTER_HOST'
```
When I bypass it by comment, I will receive another error when try to train it after load the data and model structure successfully (it halted at `line 76` of `train.py` script.):
```
Traceback (most recent call last):
  File "/home/yhqian/archives/bevfusion/tools/train.py", line 91, in <module>
    main()
  File "/home/yhqian/archives/bevfusion/tools/train.py", line 80, in main
    train_model(
  File "/home/yhqian/archives/bevfusion/mmdet3d/apis/train.py", line 49, in train_model
    model = MMDistributedDataParallel(
  File "/home/yhqian/anaconda3/envs/BEVFusionEnv/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 530, in __init__
    self.process_group = _get_default_group()
  File "/home/yhqian/anaconda3/envs/BEVFusionEnv/lib/python3.8/site-packages/torch/distributed/distributed_c10d.py", line 410, in _get_default_group
    raise RuntimeError(
RuntimeError: Default process group has not been initialized, please make sure to call init_process_group.
```
I speculate it is a issue about the normalization method (we can only use `sync_bn` in multiple GPUs?), but I have no idea how to fix it?
That is : 
1). How to train it bug-free over one GPU, where I achieve evaluate pre-trained models follow the same pipeline?
2). How to train it successfully over multiple GPUs, where I have 2 `NVIDIA 3090`, and which points I missed?
I follow the Prerequisites:
```
~ ❯ conda list | grep -E "python|openmpi|mpi4py|pillow|pytorch|mmcv|mmdet"                     🐍 BEVFusionEnv with yhqian@x86_64-conda-linux-gnu at 15:13:41
ffmpeg                    4.3                  hf484d3e_0    pytorch
ipython                   8.6.0                    pypi_0    pypi
ipython-genutils          0.2.0                    pypi_0    pypi
mmcv-full                 1.4.2                    pypi_0    pypi
mmdet                     2.20.0                    dev_0    <develop>
mmdet3d                   0.0.0                     dev_0    <develop>
mpi                       1.0                     openmpi    conda-forge
mpi4py                    3.1.3            py38h64caa1c_1    conda-forge
opencv-python             4.6.0.66                 pypi_0    pypi
openmpi                   4.1.4                external_1    conda-forge
pillow                    8.4.0                    pypi_0    pypi
python                    3.8.0                h0371630_2
python-dateutil           2.8.2                    pypi_0    pypi
python_abi                3.8                      2_cp38    conda-forge
pytorch                   1.10.2          py3.8_cuda11.3_cudnn8.2.0_0    pytorch
pytorch-cuda              11.6                 h867d48c_0    pytorch
pytorch-mutex             1.0                        cuda    pytorch
torchaudio                0.10.2               py38_cu113    pytorch
torchvision               0.11.3               py38_cu113    pytorch
```
There is nothing special for the parameter I pass, configuration file is `configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml`
Can you provide me with some insights for trouble-shooting?
Best,
Yuhang

from-Pb2Ag, 2022-11-21:
> Hi Yuhang,
> 
> Have you tried installing `torchpack` on your machine? Usually this package will help you deal with distributed training automatically, what you need to do is to add
> 
> `torchpack dist-run -np [number of gpus]`
> 
> before your normal single-GPU script, for example `python tools/train.py [config file]`. Note that even if you have just 1 GPU, specifying `-np 1` in the script will work. I think this is the easiest solution. If that does not work, let's have more dicussions under this thread.
> 
> Best, Haotian

Ok, thanks for the quick reply, I will verify it latter.

from-Pb2Ag, 2022-11-22:
> 

I follow the instruction: `torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml` and end with such error:
```
Traceback (most recent call last):
  File "tools/train.py", line 91, in <module>
    main()
  File "tools/train.py", line 80, in main
    train_model(
  File "/home/yhqian/archives/bevfusion/mmdet3d/apis/train.py", line 94, in train_model
    cfg.lr_config,
  File "/home/yhqian/anaconda3/envs/BEVFusionEnv/lib/python3.8/site-packages/mmcv/utils/config.py", line 510, in __getattr__
    return getattr(self._cfg_dict, name)
  File "/home/yhqian/anaconda3/envs/BEVFusionEnv/lib/python3.8/site-packages/mmcv/utils/config.py", line 48, in __getattr__
    raise ex
AttributeError: 'ConfigDict' object has no attribute 'lr_config'
```
which points to this block in `./mmdet3d/apis/train.py`:
```
  # register hooks
  runner.register_training_hooks(
      cfg.lr_config,
      optimizer_config,
      cfg.checkpoint_config,
      cfg.log_config,
      cfg.get("momentum_config", None),
  )
```
While I didn't **erase** any <K, V> pair in config file beside **add** some, what should I do? Wait for your reply, thanks.

from-Pb2Ag, 2022-11-22:
> I follow the instruction: `torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml` and end with such error:
>
> ```
> Traceback (most recent call [...]

This is my `torchpack` version:
```
~/ar/bevfusion on main !24 ?6 ❯ conda list | grep -E "torchpack"               🐍 BEVFusionEnv with yhqian@x86_64-conda-linux-gnu at 09:51:15
torchpack                 0.3.1                    pypi_0    pypi
```

kentang-mit, 2022-11-22:
Are you using the latest version of our codebase? The problem related to `lr_config` is not expected to appear in the latest main branch. You can add it like what I did [here](https://github.com/mit-han-lab/bevfusion/blob/f39a4a0752fabc1eb81011b0433af69a6e9ff58c/configs/nuscenes/det/transfusion/secfpn/default.yaml#L43) to your configuration file.

from-Pb2Ag, 2022-11-22:
> Are you using the latest version of our codebase? The problem related to `lr_config` is not expected to appear in the latest main branch. You can add it like what I did [...]

Hi Haotian, perhaps I shoot it out why I failed. Exactly I clone it to my laptop in the last few days of `July`, and then sync with the server rather than directly `git clone` from web, where you keep updating. I'm sorry for fail to realize it at the time when I browsing other related threads. Beside this trouble, I downgrade `setuptools` from `61.2.0` to `59.5.0`, to slove this error successfully:
```
AttributeError:module 'distutils' has no attribute 'version
```
And now I can use `torchpack` with multiple GPUs specified successfully. And I have one last question: For **train** task, how much time it about to take, before we can see something like a **ProgressBar**(just like what we can see where pickling/inferring the data.)? I kill the process for fail to see some response after run by **50min**. Suppose the same condition you noted in [issue 192](https://github.com/mit-han-lab/bevfusion/issues/192)(*i.e.*, 8xA6000, 35hs), just for reference. Thanks for your insightful instructions which helps me a lot again.

Best,
YuHang

kentang-mit, 2022-11-22:
We actually do not have progress bars during training. Instead there should be output log telling you the estimate time to finish. If you are not seeing that, I'd recommend you add a `-v` flag to the torchpack command to see whether there are error messages. Usually you should be able to see some logging message within 5 minutes.

from-Pb2Ag, 2022-11-23:
> We actually do not have progress bars during training. Instead there should be output log telling you the estimate time to finish. If you are not seeing that, I'd recommend you add a `-v` flag to the torchpack command to see whether there are error messages. Usually you should be able to see some logging message within 5 minutes after you start training.

Ok, now I can see the `.log.json` file in folder. This issue can be closed for the problem is solved successfully. Thanks for your warmhearted help.

Best,
YuHang

kentang-mit, 2022-11-23:
Glad to hear that! I'll close this issue for now.

closed 2022-11-23 by kentang-mit