mit-han-lab/bevfusion #241: Error while training the fusion model

oliver0922, 2022-11-23:
Hi, Thanks for your wonderful work!
I am trying to train the bevfusion model and the command was below.

`torchpack dist-run -np 4 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml
`
However the error occured like this
[image]

How can I solve this error?

VeeranjaneyuluToka, 2022-11-25:
Would you mind mentioning your gpu card details here? I suspect it is due to insufficient GPU memory.

oliver0922, 2022-11-26:
@VeeranjaneyuluToka 
Thanks for your reply but I don't think that the problem is related to GPU memory.
I used 4 RTX 3090 GPUs.

kentang-mit, 2022-12-12:
Sorry I missed the issue @oliver0922. It could be the GPU memory issue. Our fusion model is expected to consume >24G GPU memory. You might need to reduce the batch size (`data.samples_per_gpu`).

kentang-mit, 2022-12-27:
Closed due to inactivity.

closed 2022-12-27 by kentang-mit