mit-han-lab/bevfusion #242: Performance on nuscenes test

aurorasyu, 2022-11-24:
Hi, thanks for your excellent work.

I have some questions about the result on nusenes test set. Our result of test set trained from train set is similar to the result from train+val set. But you result of the test is 2% mAP when train on train+val. Is there anything different when train on train+val.

Thanks for your help.

sandropapais, 2022-11-24:
I have a similar issue. When I evaluate the pretrained model provided the performance is mAP=0.6846 and NDS=0.7143 (same as the paper). However, when I train the model from scratch using the provided config and evaluate it, the performance is mAP=0.6440 and NDS=0.6933. So there is a gap of 4% mAP and 2% NDS when I try to reproduce the results.

Is there a different config file or an older version that should be used to reproduce the results?

kentang-mit, 2022-11-25:
We use the same schedule for models on val and models on test. The only difference is that we use the training set for models evaluated on the validation set and we use train+val set for models evaluated on the test set. As far as I know, CenterPoint, TransFusion and our method all have slightly higher test mAP (compared with val mAP), but the performance difference varies from model to model.

aurorasyu, 2022-12-08:
The Lidar pretrain model also use train+val? or only fusion use train+val?
when i use the train+val to train the lidar model, I find the performance on truck has twenty points lower campare to train on train set.

aurorasyu, 2022-12-09:
Thanks for your answer! 
Another question is that the train+val you used is combine the train.pkl+val.pkl during create data?

kentang-mit, 2022-12-10:
@aurorasyu, Yes, you are right.

kentang-mit, 2022-12-27:
Closed due to inactivity.

closed 2022-12-27 by kentang-mit