mit-han-lab/bevfusion #243: About training

YangChen1234567, 2022-11-25:
Hi~
Thanks for your nice work.... But when reproducing the results, i met some problems.

I used the training instructions as below to train fusion model(diden't load pretrained lidar model), 
`torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml`
but my val mAP is always zero, like this:(scalars from tensorboard, trained after 5 epochs)
[image]

Is it normal? Is there any details when training that i dont'know??  
This is the traing loss:
[image]

YangChen1234567, 2022-11-25:
and my batchsize is 2 (2 GPUS, samples_per_gpu = 1)

kentang-mit, 2022-12-27:
Hi @YangChen1234567,

Sorry for missing that issue. I believe you should use the training command for the fusion model provided by us in the current main branch.

Best,
Haotian

closed 2023-03-30 by kentang-mit