mit-han-lab/bevfusion #244: Question about a data augmentation

xxxxhh, 2022-11-25:
Hi, I would like to make a check of whether the GridMask augmentation is used to random mask the grids in the image plane or not. By the way, have you studies the influence of this data augmentation> Thanks!

kentang-mit, 2022-11-25:
GridMask is not used in any of the results reported in our arXiv version. However, I remember that it works for some cases (e.g. sparse LiDAR + camera).

xxxxhh, 2022-11-26:
Got that，thanks! So，if I would like to reproduce the results on ArXiv，I should set the ratio parameters in the GridMask as 0 in the configs. Do I understand your points right?

kentang-mit, 2022-11-27:
Yes, you are correct.

closed 2022-11-28 by xxxxhh

liyih, 2023-10-17:
@xxxxhh ，hi, I am wondering have you done the experiment. Whether GridMask will improve the performance of BEVFusion or Not.
Best.