mit-han-lab/bevfusion #246: About visualization frame numbers

derekray311511, 2022-11-27:
Hi! Thanks for the excellent work.
I'm trying to visualize the result from bevfusion.
It worked with this command but the results images from camera and lidar have only 5973 frames, not 6019 as I expected, do you know why this happened? 
$"torchpack dist-run -np 1 python tools/visualize.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --checkpoint pretrained/bevfusion-det.pth --out-dir /dataset/viz/fusion-det-pred --mode pred --bbox-score 0.1"
Many thanks!

kentang-mit, 2022-11-27:
That's interesting. I remember that we saw similar problems before and at that time we did not figure out the reason. But there is one solution if you are strict about getting 6019 samples: instead of iterating over the validation loader [here](https://github.com/mit-han-lab/bevfusion/blob/da751a2c9fa7d436f4ba22427ded5782f3b9a50e/tools/visualize.py#L82), you could iterate over the entries info pkl file.

derekray311511, 2022-11-28:
Do you mean load the data from pkl file directly but not from dataloader and dataset? By the way, I found something new and still trying to figure out why this happened. The dataloader have the correct len 6019, but sometimes it will load the same token with same timestamp, which overwrite the previous frame, and finally cause result losing frames.
[image]

kentang-mit, 2022-11-28:
Hi @derekray311511,

Yes, previously I mean you can directly get the data infos from the pkl file. But since you have found out that the dataloader has correct length, it seems that the problem you met is a naming issue. In this case, you may change the naming convention (e.g. use the frame index as the filename). 

Best,
Haotian

derekray311511, 2022-11-28:
Thank you for your response! The problem is solved. Changing the names works and I also found that changing the val `test_mode` to True in the config file also works. I think the val `test_mode` default by False is used to train the model for test data.

closed 2022-11-29 by kentang-mit