mit-han-lab/bevfusion #248: The program seems to stuck

LeongVan, 2022-11-28:
Hi, thanks for opening source program. I've got a question. 
I was running the evaluation using command `CUDA_VISIBLE_DEVICE=7 torchpack dist-run -np 4 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox` (CUDA_VISIBLE_DEVICE=7 is used to control which GPU device is activated, '7' means I merely used the 8th one). But I got the following situation: the program seems to stuck without any process running. I've been waiting for half an hour and no process was going on. Is it normal or should I waiting for more time? The print-out in terminal is as below. 
```
(BEVFusion385) root@ubuntu:/sdd1/bevfusion# CUDA_VISIBLE_DEVICE=7 torchpack dist-run -np 4 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
[ubuntu:2322777] pmix_mca_base_component_repository_open: unable to open mca_pnet_test: perhaps a missing symbol, or compiled for a different version of OpenPMIx (ignored)
[ubuntu:2322777] pmix_mca_base_component_repository_open: unable to open mca_pnet_tcp: perhaps a missing symbol, or compiled for a different version of OpenPMIx (ignored)
Config (path: configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml): {'seed': 0, 'deterministic': False, 'checkpoint_config': {'interval': 1, 'max_keep_ckpts': 1}, 'log_config': {'interval': 50, 'hooks': [{'type': 'TextLoggerHook'}, {'type': 'TensorboardLoggerHook'}]}, 'load_from': None, 'resume_from': None, 'cudnn_benchmark': False, 'fp16': {'loss_scale': {'growth_interval': 2000}}, 'max_epochs': 20, 'runner': {'type': 'CustomEpochBasedRunner', 'max_epochs': 20}, 'dataset_type': 'NuScenesDataset', 'dataset_root': 'dataset/NuScenes/', 'gt_paste_stop_epoch': -1, 'reduce_beams': 32, 'load_dim': 5, 'use_dim': 5, 'load_augmented': None, 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0], 'voxel_size': [0.075, 0.075, 0.2], 'image_size': [256, 704], 'augment2d': {'resize': [[0.38, 0.55], [0.48, 0.48]], 'rotate': [-5.4, 5.4], 'gridmask': {'prob': 0.0, 'fixed_prob': True}}, 'augment3d': {'scale': [0.9, 1.1], 'rotate': [-0.78539816, 0.78539816], 'translate': 0.5}, 'object_classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'map_classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider'], 'input_modality': {'use_lidar': True, 'use_camera': True, 'use_radar': False, 'use_map': False, 'use_external': False}, 'train_pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 9, 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'pad_empty_sweeps': True, 'remove_close': True, 'load_augmented': None}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True, 'with_attr_label': False}, {'type': 'ObjectPaste', 'stop_epoch': -1, 'db_sampler': {'dataset_root': 'dataset/NuScenes/', 'info_path': 'dataset/NuScenes/nuscenes_dbinfos_train.pkl', 'rate': 1.0, 'prepare': {'filter_by_difficulty': [-1], 'filter_by_min_points': {'car': 5, 'truck': 5, 'bus': 5, 'trailer': 5, 'construction_vehicle': 5, 'traffic_cone': 5, 'barrier': 5, 'motorcycle': 5, 'bicycle': 5, 'pedestrian': 5}}, 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'sample_groups': {'car': 2, 'truck': 3, 'construction_vehicle': 7, 'bus': 4, 'trailer': 6, 'barrier': 2, 'motorcycle': 6, 'bicycle': 6, 'pedestrian': 2, 'traffic_cone': 2}, 'points_loader': {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32}}}, {'type': 'ImageAug3D', 'final_dim': [256, 704], 'resize_lim': [0.38, 0.55], 'bot_pct_lim': [0.0, 0.0], 'rot_lim': [-5.4, 5.4], 'rand_flip': True, 'is_train': True}, {'type': 'GlobalRotScaleTrans', 'resize_lim': [0.9, 1.1], 'rot_lim': [-0.78539816, 0.78539816], 'trans_lim': 0.5, 'is_train': True}, {'type': 'LoadBEVSegmentation', 'dataset_root': 'dataset/NuScenes/', 'xbound': [-50.0, 50.0, 0.5], 'ybound': [-50.0, 50.0, 0.5], 'classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider']}, {'type': 'RandomFlip3D'}, {'type': 'PointsRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectNameFilter', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'ImageNormalize', 'mean': [0.485, 0.456, 0.406], 'std': [0.229, 0.224, 0.225]}, {'type': 'GridMask', 'use_h': True, 'use_w': True, 'max_epoch': 20, 'rotate': 1, 'offset': False, 'ratio': 0.5, 'mode': 1, 'prob': 0.0, 'fixed_prob': True}, {'type': 'PointShuffle'}, {'type': 'DefaultFormatBundle3D', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'Collect3D', 'keys': ['img', 'points', 'gt_bboxes_3d', 'gt_labels_3d', 'gt_masks_bev'], 'meta_keys': ['camera_intrinsics', 'camera2ego', 'lidar2ego', 'lidar2camera', 'camera2lidar', 'lidar2image', 'img_aug_matrix', 'lidar_aug_matrix']}], 'test_pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 9, 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'pad_empty_sweeps': True, 'remove_close': True, 'load_augmented': None}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True, 'with_attr_label': False}, {'type': 'ImageAug3D', 'final_dim': [256, 704], 'resize_lim': [0.48, 0.48], 'bot_pct_lim': [0.0, 0.0], 'rot_lim': [0.0, 0.0], 'rand_flip': False, 'is_train': False}, {'type': 'GlobalRotScaleTrans', 'resize_lim': [1.0, 1.0], 'rot_lim': [0.0, 0.0], 'trans_lim': 0.0, 'is_train': False}, {'type': 'LoadBEVSegmentation', 'dataset_root': 'dataset/NuScenes/', 'xbound': [-50.0, 50.0, 0.5], 'ybound': [-50.0, 50.0, 0.5], 'classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider']}, {'type': 'PointsRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ImageNormalize', 'mean': [0.485, 0.456, 0.406], 'std': [0.229, 0.224, 0.225]}, {'type': 'DefaultFormatBundle3D', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'Collect3D', 'keys': ['img', 'points', 'gt_bboxes_3d', 'gt_labels_3d', 'gt_masks_bev'], 'meta_keys': ['camera_intrinsics', 'camera2ego', 'lidar2ego', 'lidar2camera', 'camera2lidar', 'lidar2image', 'img_aug_matrix', 'lidar_aug_matrix']}], 'data': {'samples_per_gpu': 4, 'workers_per_gpu': 4, 'train': {'type': 'CBGSDataset', 'dataset': {'type': 'NuScenesDataset', 'dataset_root': 'dataset/NuScenes/', 'ann_file': 'dataset/NuScenes/nuscenes_infos_train.pkl', 'pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 9, 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'pad_empty_sweeps': True, 'remove_close': True, 'load_augmented': None}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True, 'with_attr_label': False}, {'type': 'ObjectPaste', 'stop_epoch': -1, 'db_sampler': {'dataset_root': 'dataset/NuScenes/', 'info_path': 'dataset/NuScenes/nuscenes_dbinfos_train.pkl', 'rate': 1.0, 'prepare': {'filter_by_difficulty': [-1], 'filter_by_min_points': {'car': 5, 'truck': 5, 'bus': 5, 'trailer': 5, 'construction_vehicle': 5, 'traffic_cone': 5, 'barrier': 5, 'motorcycle': 5, 'bicycle': 5, 'pedestrian': 5}}, 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'sample_groups': {'car': 2, 'truck': 3, 'construction_vehicle': 7, 'bus': 4, 'trailer': 6, 'barrier': 2, 'motorcycle': 6, 'bicycle': 6, 'pedestrian': 2, 'traffic_cone': 2},  [... cut ...]
[  ] 4/6019, 0.2 task/s, elapsed: 19s, ETA: 27922s/sdd1/bevfusion/mmdet3d/core/bbox/coders/transfusion_bbox_coder.py:96: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  self.post_center_range = torch.tensor(
[>>] 6020/6019, 8.6 task/s, elapsed: 700s, ETA:     0s
```

When I watch my GPU device, instead of the 8th one, the device activated actually are 1st, 3rd and 4th one. All of them are 100% used while the program seems to stuck.

LeongVan, 2022-11-28:
It works now. I just simply stop and re-run it for several times, and then the program works.

zhanglk9, 2022-12-02:
> 现在可以了。我只是简单地停止并重新运行它几次，然后程序就可以运行了。

I rerun many times, but it doesn't solve the problem

zhanglk9, 2022-12-02:
I meet the same problem. Do you solve that now??

LeongVan, 2022-12-05:
> I meet the same problem. Do you solve that now??

Yes, just simply rerun it. Nothing else has changed.

lilkeker, 2022-12-05:
I also encountered the same problem. I modified part of the code in LSS, and the debugging can be successful, but it will get stuck at this place. If you use the original code that has not been modified, everything is normal

lilkeker, 2022-12-06:
> I meet the same problem. Do you solve that now??
Hello, did you solve this problem？ I wonder if this is a bug

zhanglk9, 2022-12-06:
> > I meet the same problem. Do you solve that now??
> > Hello, did you solve this problem？ I wonder if this is a bug

Today I use one GPU to test ,and I didn't meet this problem.But using 4 GPUs，it doesn't work.

LeongVan, 2022-12-07:
> I also encountered the same problem. I modified part of the code in LSS, and the debugging can be successful, but it will get stuck at this place. If I use the original code that has not been modified, everything is normal

I didn't changed the original code, but still met the problem. Yesterday, I trained on 4 GPUs, and when I checked out this morning, it stuck again.

lilkeker, 2022-12-07:
> I didn't changed the original code, but still met the problem. Yesterday, I trained on 4 GPUs, and when I checked out this morning, it stuck again.
I'm wondering if it's a problem with the environment installation. This is the process of my environment installation. If I don't modify any code, there will be no problem running. You can refer to it
[image]

lilkeker, 2022-12-07:
> Hi, thanks for opening source program. I've got a question. I was running the evaluation using command `CUDA_VISIBLE_DEVICE=7 torchpack dist-run -np 4 python tools/test.py [...]
Did you make a mistake here? Your code is for a single GPU, so it should be "np 1"

LeongVan, 2022-12-07:
> 

Thanks for your kind reply. Yes, that's a mistake. Now I notice that. I'll try your instruction to install a new environment to see whether everything can work as expected.

LeongVan, 2022-12-07:
> I also encountered the same problem. I modified part of the code in LSS, and the debugging can be successful, but it will get stuck at this place. If I use the original code that has not been [...]

Thanks a lot! 

I set up a brand new environment under your instruction, and it works except a little error when running `python setup.py develop` as `AttributeError: module 'distutils' has no attribute 'version'`. 

To solve this, I follow [the solution](https://github.com/pytorch/pytorch/issues/69894) here, which is to `pip install setuptools==59.5.0`, and then run setup.py and everything works!

lilkeker, 2022-12-09:
I have not encountered a similar problem. I run the camera plus lidar detection test now. There is no problem. I think this is caused by a wrong environment installation. You can try "pip install mmcv-full==1.4.0 - f https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html", because I see that you are not the same as me when installing mmcv, if not, I don’t know how to solve it up

LeongVan, 2022-12-09:
Thanks for your advice. I have tried both `pip install mmcv-full==1.4.0` and `pip install mmcv-full==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html`. Unfortunately, none of them works. Same bug appears.

lilkeker, 2022-12-09:
> Thanks for your advice. I have tried both `pip install mmcv-full==1.4.0` and `pip install mmcv-full==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html`. Unfortunately, none of them works. Same bug appears.

Sorry, I can't solve the problem you are currently encountering, if you solve it, please let me know your solution

LeongVan, 2022-12-09:
> 

Now the problem is more clear. 
If I run on single one or two GPUs, the program runs successfully. Instead, if I run on more than 2 GPUs, such as 3, 4, or more, the problem mentioned above is triggered. Such a strange bug.

kentang-mit, 2022-12-12:
Sorry for missing the discussion. I think you might want to check whether you used some flags like `CUDA_VISIBLE_DEVICES` in your command. If you specified that only two GPUs are visible and you want to use `-np 4` it might not work.

LeongVan, 2022-12-13:
> Sorry for missing the discussion. I think you might want to check whether you used some flags like `CUDA_VISIBLE_DEVICES` in your command. If you specified that only two GPUs are visible and you want to use `-np 4` it might not work.

Thanks for your reply. 
Actually I've noticed that little typo already, so the running cammands afterwards I used is something like `CUDA_VISIBLE_DEVICES=1,2,3 torchpack dist-run -np 3 -v python tools/train.py configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yam`, where `-np` is consistent with the number of GPUs I used. But the same bug appear as well, as printed in the terminal, `N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0`. 
Also, I checked the discussion in [#28](https://github.com/mit-han-lab/bevfusion/issues/28), but found no solution to this bug. It seems like a problem related to spconv block. 
Any clue will be appreciate.

kentang-mit, 2022-12-13:
Did you have a chance to check out the answer in https://github.com/mit-han-lab/bevfusion/issues/28#issuecomment-1160333858? It seems that you are using CUDA 11.7, but you are trying to configure the CUDA environment in conda to be 11.3. I can confirm that the current version of this codebase requires the PyTorch version to be 1.10.x, and probably there is no CUDA11.7-compiled PyTorch wheel available. So my suggestion will be to install cuda-11.3 on your local machine, temporarily symbolic link `/usr/local/cuda` to `/usr/local/cuda-11.3` and try again. 

By the way, we are currently working on a new version of this codebase without the spconv backend. It will take some time, please stay tuned.

LeongVan, 2022-12-13:
> 

Thanks for your suggestion. 
I tried CUDA 11.3 as well, but it didn't work... The CUDA in system is 11.7 (base environment), and I installed CUDA 11.3 in my virtual environment. 
[image]

kentang-mit, 2022-12-14:
Yes, I understand your problem, and I just want to confirm whether you tried to install CUDA 11.3 on the ubuntu system (not in the anaconda virtual environment). According to the discussions in issue 28, it seems that your CUDA version in the OS should match with the CUDA version in anaconda. 

I knew that the environment setup could be painful, and we are working hard to make it easier.

lilkeker, 2022-12-23:
我仍然遇见这个问题，今天我在跑程序的时候，它又卡住了，
[image]

###############################################################
(packnetsan) lk@user-SYS-7049GP-TRT:~/bevfusion_ori$ nvidia-smi
Fri Dec 23 13:06:16 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 465.19.01    Driver Version: 465.19.01    CUDA Version: 11.3     |
|-------------------------------+----------------------+----------------------+

#################################################################
packnetsan) lk@user-SYS-7049GP-TRT:~/bevfusion_ori$ python
Python 3.8.13 (default, Oct 21 2022, 23:50:54) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> torch.__version__
'1.10.0+cu113'
>>> torch.version.cuda
'11.3'
>>> torch.backends.cudnn.version()
8200
>>> exit()

kentang-mit, 2022-12-23:
Hi @lilkeker,

Did you have a chance to add a `-v` flag to the command you are running and see if there will be any error messages? 

Best,
Haotian

lilkeker, 2022-12-30:
> Hi @lilkeker,
> 
> Did you have a chance to add a `-v` flag to the command you are running and see if there will be any error messages?
> 
> Best, Haotian
Thanks a lot ,Haotian, I think I have found the reason, it is because I modified a part of the code,and the code will be stuck
`  File "tools/test.py", line 205, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/mmdet/apis/test.py", line 98, in multi_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/lk/bevfusion_ori/mmdet3d/models/fusion_models/bevfusion.py", line 188, in forward
    outputs = self.forward_single(
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/lk/bevfusion_ori/mmdet3d/models/fusion_models/bevfusion.py", line 231, in forward_single
    feature = self.extract_camera_features(
  File "/home/lk/bevfusion_ori/mmdet3d/models/fusion_models/bevfusion.py", line 114, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lk/bevfusion_ori/mmdet3d/models/vtransforms/depth_lss.py", line 260, in forward
    x = super().forward(*args, **kwargs)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/lk/bevfusion_ori/mmdet3d/models/vtransforms/base.py", line 293, in forward
    x = self.get_cam_feats(img, depth, raw_img)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/lk/bevfusion_ori/mmdet3d/models/vtransforms/depth_lss.py", line 128, in get_cam_feats
    d_fill = packnetsanpre(raw_img,d)
  File "/home/lk/bevfusion_ori/packnetmain/scripts/test_depth_net.py", line 61, in packnetsanpre
    pred_inv_depth = model_wrapper.depth(inputrgb,input_depth)['inv_depths'][0]
  File "/home/lk/bevfusion_ori/packnetmain/packnet_sfm/models/model_wrapper.py", line 284, in depth
    return self.depth_net(*args, **kwargs)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lk/bevfusion_ori/packnetmain/packnet_sfm/networks/depth/PackNetSAN01.py", line 220, in forward
    inv_depths_rgb, skip_feat_rgb = self.run_network(rgb)
  File "/home/lk/bevfusion_ori/packnetmain/packnet_sfm/networks/depth/PackNetSAN01.py", line 197, in run_network
    x5p, skips = self.encoder(rgb)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lk/bevfusion_ori/packnetmain/packnet_sfm/networks/depth/PackNetSAN01.py", line 34, in forward
    x = self.pre_calc(rgb)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lk/bevfusion_ori/packnetmain/packnet_sfm/networks/layers/packnet/layers01.py", line 37, in forward
    x = self.conv_base(self.pad(x))
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/conv.py", line 446, in forward
    return self._conv_forward(input, self.weight, self.bias)
  File "/home/lk/anaconda3/envs/packnetsan/lib/python3.8/site-packages/torch/nn/modules/conv.py", line 442, in _conv_forward
    return F.conv2d(input, weight, bias, self.stride,
RuntimeError: Expected all tensors to be on the same device, but found at least two devices, cuda:1 and cuda:0! (when checking argument for argument weight in method wrapper__cudnn_convolution)`

kentang-mit, 2022-12-31:
Hi @lilkeker,

It seems to me that there are some custom modules that are not implemented by us in your codebase. What the error means  is that the input and weight to your convolution layer (`self.conv_base` in the error log) are not on the same GPU. You may need to check what happened in this function call in your custom modules.

Best,
Haotian

kentang-mit, 2023-03-30:
Closed due to inactivity. Please feel free to reopen if you feel it necessary.

closed 2023-03-30 by kentang-mit

ZoeSnow, 2023-12-13:
I find a solution and it works, maybe you can try it. Reference https://zhuanlan.zhihu.com/p/319888724, it is caused by the types of CPU and GPU