mit-han-lab/bevfusion #249: The programs will be stuck when training the lidar-only model

ChongjianGE, 2022-11-29:
Hi, 

Thanks for the wonderful project. I encounter some problems when training the lidar-only model. 
Specifically, the program will be stuck randomly during the training process. When the programs are stuck, some of the GPU-Util will be 0%, while others are 100%, which is shown in the following picture. I have tried to reduce the batchsize (since I guess there might be some OOM errors), but it would not solve the problems. 

Are there any suggestions that you could provide?

Best
[image]

LeongVan, 2022-11-29:
I have the same question with you when training LiDAR-only model. I was stop here: 
[image]
I used 3 GPUs. When stucking, 2 of them are 100% used and the last one is 0%.

sihaoevery, 2022-12-04:
Hi everyone,

Just encountered the same issue when training the lidar-only detection.
I use a cluster and submit the job by slurm.  I have 4 A100, 48 CPU cores, and 64G memory. Both the workers_per_gpu and samples_per_gpu are set to 8. The printed log says it will take ~40h. 

Everything seems to be smooth before it comes to epoch 8.  One GPU utility is 0% while others are 100%. Then I tried to resume the model from epoch 7 and it just stuck again in epoch 8.  Nothing is suggested in both the slurm output or error log.

Now I try to train the lidar-camera detection to see if this problem still persists.

Any suggestion would be appreciated.

Thanks in advance

LeongVan, 2022-12-05:
My problem solved just simply re-run the program...Seems strange, but it works. I've tried many times until it works.

sihaoevery, 2022-12-05:
> 

Thanks for this information.

Can you provide more information? By 're-run', do you mean train from scratch or resume the model? And what's the setting of batch size and workers for dataloader?

Many thanks.

LeongVan, 2022-12-05:
> > 
> 
> Thanks for this information.
> 
> Can you provide more information? By 're-run', do you mean train from scratch or resume the model? And what's the setting of batch size and workers for dataloader?
> 
> Many thanks.

I trained from scratch because I was stuck at only epoch 1. I didn't change batch size and workers, so they are default. I checked out that workers is 4 and I can't found the batch size parameter. I'll appreciate if you could guide me where the batch size is set. 
I ran my program on 3 Tesla A100 with 80GB each of them. And the usage of them when training were close to 100% all the training time.

LiewFeng, 2023-04-04:
I get stuck during the training of lidar only model, too. @zhijian-liu ,could you kindly give any suggestions? Lots of thanks.

LiewFeng, 2023-04-04:
I use 8 A100 and prepare the enviroment as the readme suggests and don't modify any hyper-parameters. I tried twice but get stuck in the same position.
[image]

VeeranjaneyuluToka, 2023-04-20:
Are you trying with custom dataset or Nuscenes dataset, i have trying with custom dataset, and ended with the error mentioned in the below issue. https://github.com/mit-han-lab/bevfusion/issues/382, anybody faced this issue and fixed? Thanks!

pianogGG, 2023-07-06:
I got the same issue， everything seems to be smooth before it comes to epoch 5. One GPU utility is 0% while others are 100%. Good idea? thanks a lot.

Ricardolyg, 2023-08-07:
change the batch_size in bevfusion/configs/nuscenes/default.yaml
samper per_gpu : 1
worker per_gpu : 1

closed 2024-05-04 by zhijian-liu