mit-han-lab/bevfusion #250: the problem of torchpack export onnx

yzjalien, 2022-11-29:
when i want to export onnx ,the error as follows:
what`s the problem of this?
[image]

thanks

kentang-mit, 2022-11-29:
Hi @yzjalien,

Onnx export might be very challenging since sparse convolution operators are not natively supported. This means that we have to register custom ops in onnx. We'll try to support that in the future, but in the near term this is hard.

If you want to do TensorRT latency measurement, I would recommend you to export those 2D-only parts separately.

Best,
Haotian

yzjalien, 2022-11-30:
> Hi @yzjalien,
>
> Onnx export might be very challenging since sparse convolution operators are not natively supported. This means that we have to register custom ops in onnx. We'll try to support that [...]

thanks for your response,and i will be appreciate if i know how to export 2D-only parts ?

kentang-mit, 2022-11-30:
I think you probably need to change the code a little bit. After creating the full model, you could access the 2D-only parts using `model.fuser` and  `model.decoder`. You could then create dummy input tensors with correct input shapes (e.g. inputs to the fuser should be 1x256x180x180 and 1x80x180x180) to these modules and export the onnx. `model.encoders["camera"]` (except the vtransform module) is also directly exportable.