mit-han-lab/bevfusion #251: the problem of training

yzjalien, 2022-11-30:
first, run:
torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml
then ,the errors are as follows:
[image]
here is the cuda information:
[image]
is  GPU memory not enough? if it is true, how to cut down the training memory?

kentang-mit, 2022-11-30:
No it might not be enough for training. I think you can definitely use a smaller batch size, but I think 8G will probably be not enough for fusion model training even if you use batch size = 1.

yzjalien, 2022-12-01:
understood, it failed while i use batch_size to 1.
It seems to expand my gpu memory .
thanks~

kentang-mit, 2022-12-01:
I'm closing this issue since it is resolved. Feel free to reopen if you have further questions.

closed 2022-12-01 by kentang-mit