mit-han-lab/bevfusion #253: Nuscenes validation detection performance

sandropapais, 2022-12-01:
I am not able to reproduce the nuscenes validation dataset performance that was reported. I am training on 8xV100 GPU with the provided config in the convfuser.yaml file.

When I evaluated the pre-trained model the performance is mAP: 0.6846, NDS: 0.7143 as reported. But when I trained the model from scratch the performance is only mAP: 0.6440, NDS: 0.6933. I also tried training with the pretrained lidar weights (using `--load_from pretrained/lidar-only-det.pth`) as mentioned in #207 and the performance is only mAP: 0.6752, NDS: 0.7121. 

There is still a gap of about 1% in mAP. Is this to be expected or is there a way to address this in training?

YangChen1234567, 2022-12-02:
Hi, when you trained with the pretrained lidar weights, have you found the validation map and nds values was getting lower and lower after each epoch? I meet this problem, but i don't know why....

kentang-mit, 2022-12-04:
Please use a cosine annealing learning rate schedule with starting learning rate = 2e-4 and train for six epochs. Loading pretrained LiDAR weights is necessary. Also, please remember to run the test.py script separately after the training. I'll release the configs very soon.

kentang-mit, 2022-12-27:
Closed due to inactivity.

closed 2022-12-27 by kentang-mit