mit-han-lab/bevfusion #255: The problem about evaluating the pretrained models

zhanglk9, 2022-12-02:
I use 4 GPUs(GeForce RTX 3090) to evaluate the the pretrained model by runing the follow:
`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`
but the  process always gets stuck and doesn't get the error:
[image]
And I saw that just 0s,1s and 3s GPU was been used.I trid several time but got the same problem.

zhanglk9, 2022-12-02:
[image]
GPU performance as above

kentang-mit, 2022-12-04:
In this case it seems that you should not use `-np 8` because you have only 4 GPUs.

zhanglk9, 2022-12-05:
> In this case it seems that you should not use `-np 8` because you have only 4 GPUs.
Sorry, I made a typo here. I used '-np 4' for testing, and I also tried '-np 1' and '-np 3''-np2',but the problem is the same. Sometimes I could meet this situation:
[image]

kentang-mit, 2022-12-05:
It seems that you did not install `mmcv-full==1.4.0` and `mmcv==1.4.0` according to our suggestions in the README.md (otherwise spconv should not be part of mmcv-full). Please follow the instructions carefully and see if you can solve the problem.

zhanglk9, 2022-12-06:
> It seems that you did not install `mmcv-full==1.4.0` and `mmcv==1.4.0` according to our suggestions in the README.md (otherwise spconv should not be part of mmcv-full). Please follow the instructions carefully and see if you can solve the problem.

https://github.com/mit-han-lab/bevfusion/issues/248#issuecomment-1339317712

I alse think testing the pretrained models with more than 1 GPU can be stuck is a bug

kentang-mit, 2022-12-07:
Would you mind providing more information about your environment? E.g. CUDA version, PyTorch version, etc. Our recommendation is to use PyTorch 1.10.x + CUDA 11.3. You may also try out our docker environment, I've tested it multiple times on different machines (with CUDA 11.3), and every time I use >1 GPUs to run evaluation and training. 

By the way, the setup instruction in the issue 248 you mentioned is very helpful.

kentang-mit, 2022-12-27:
Closed due to inactivity.

closed 2022-12-27 by kentang-mit