mit-han-lab/bevfusion #256: About tracking and coordinate system

derekray311511, 2022-12-03:
Hi! I'm here again. I was stuck by two problems these days when I try to perform tracking.  

- Inference coordinate system (Nuscenes):
Do the model inference on the LiDAR coordinates? If yes, where do the yaw angle start from?  
This is my steps to track:  
1. Get outputs from BEVfusion-det model.
2. Format the outputs to nuscenes format.
[image]
3. Transform the objects to global coordinates.
4. Perform tracking same as centerpoint.  
I think the key point is step two. Is there any wrong especially on yaw angle and velocity?  
When I visualize them, some velocity will be on the opposite like this.
[image]

- Do you use bbox threshold when performing tracking?

derekray311511, 2022-12-03:
Oh! I just fixed the first problem.  
The inference bboxes is [x, y, z, w, l, h, yaw, vx, vy] but not [x, y, z, w, h, l, yaw, b, v].
[image]

kentang-mit, 2022-12-04:
Sounds good. Good luck for your tracking experiments.

xjturjc, 2023-01-11:
@derekray311511  Hello, may I know your tracking results on nuScenes validation? I transplanted the tracker from CenterPoint, and the AMOTA is only 0.70. Thanks.

derekray311511, 2023-01-11:
@xjturjc Hi! I also used the tracker from CenterPoint. I think I got the similar result as you did by directly using the CenterPoint tracker. You can simply get a better result by filtering out the low score bounding boxes. In my case, I set the threshold to 0.025 and got the AMOTA 0.742 on the validation set.

closed 2023-03-30 by kentang-mit

changxu-zhang, 2023-11-21:
Hi @derekray311511, may I ask how you visualized tracking results from CenterPoint? Is there any off-the-shelf tools that I didn't notice? Thanks

derekray311511, 2023-11-21:
Hi, @changxu-zhang, I write the visualization code by myself. It use ROS. You can see the code in my repository here, github.com/derekray311511/nusc_tracking

changxu-zhang, 2023-11-24:
Hi @derekray311511, thanks for your share. I've also built one myself inspired by yours.