mit-han-lab/bevfusion #258: How to inference with custom image data?

LeongVan, 2022-12-05:
Thanks for nice work. 
I've finished training camera-only model using `NuScenes/v1.0-trainval` dataset, and generated a `lasted.pth`. I have 2 images and I wonder how to use this trained model to inference these 2 custom image data. Any clue?

kentang-mit, 2022-12-05:
Hi @LeongVan,

You probably need more than just 2 images as the input. Projection matrices (e.g. `camera_intrinsics`, `camera2lidar`) are also necessary. You can have a look at our argument list for the forward function [here](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/fusion_models/bevfusion.py#L166) for more details.

Best,
Haotian

kentang-mit, 2022-12-10:
@whoakang I haven't really tried inference with custom image data, so I cannot provide a code example for you to begin with. But as I have mentioned you need to provide all fields required by our forward function. If you met with any problems, feel free to continue the discussion in this thread.

LeongVan, 2022-12-20:
Hi, @kentang-mit 

I've tried to implement inference of custom data. My workflow is to replace `[data'img']` with costom data (assume other fields are the same with NuScenes because my costom images are chosen from NuScenes from 6 different cameras). I found that in `tools/visualize.py`, when loading images into the dataloader, there exists some pre-processing, right? 

I check `test_pipeline` in config file then. There are 11 types of classes, and two among them, `ImageAug3D` and `ImageNormalize`, are dealing with `data['img']`. So I tried to pre-process the images using `ImageAug3D` and `ImageNormalize`. Then I inferenced, and found that the output is very strange. I suppose that the pre-processing wasn't conducted properly, but I cannot find it. 

I want to upload the inference figure for convenient check, but that operation is forbidden caused by company's security rules. Any clue will be greatly appreciate.

kentang-mit, 2022-12-21:
Hi @LeongVan,

Yes there are some custom preprocessing to the images. I think the tricky part is how you choose these 6 images on nuScenes. Not all scenes on the nuScenes dataset have exactly the same camera intrinsic and extrinsic matrices. That said, if you choose images with different camera intrinsic / extrinsic matrices from what is specified in the forward function, the inference results could mess up. Would you mind first checking whether the original nuScenes data will work for you, and then start customization step-by-step (for example, assembling six camera inputs from adjacent frames, choosing CAM_FRONT from frame1, CAMERA_BACK from frame 2, etc.)? 

Best,
Haotian

zhiranyan, 2023-03-24:
Hi, I am just thinking that i will build my own dataset (e.g 6 different view images, 1 lidar pointcloud, camera intrinsic, extrinsic data) and dataloader (rather than pkl files as nuscenes dataset case) and feed to the algorithm, would that work? Do you have a demo for this? Just for inferencing of one frame data. Thanks!

closed 2023-07-15 by zhijian-liu