mit-han-lab/bevfusion #260: Could you share the configs for Waymo?

FrontierBreaker, 2022-12-07:
Hi, I would like to know if you could share the configs and dataloader for waymo dataset as I would like to make some simple experiments on that based on BEVFusion, thanks!

kentang-mit, 2022-12-08:
Hi @FrontierBreaker,

We are still cleanning up the code for Waymo. The dataloader implementation is going to be completely different from the one in mmdetection3d. Please stay tuned and we will keep you posted.

Best,
Haotian

FrontierBreaker, 2022-12-08:
Thanks! Looking forward to that ~

FrontierBreaker, 2022-12-21:
Hi, @kentang-mit , could you provide a rough release time of the waymo dataloader? Thanks!

Qjizhi, 2023-02-27:
> It's a bit hard to estimate the release time. We are doing the following things to this codebase:
> 
> * Reformatting the APIs such that the dependency on library (mmcv/mmdet/etc.) versions will become less strict (a lot of users are facing these issues);
> * Replacing the sparse convolution backend to solve issues related to spconv;
> * Releasing support for more datasets.
> 
> I would say that the first two things are of higher priority on our roadmap because many users encountered problems related to these two items.

@kentang-mit Hi Kentang, how is it going?
Are you going to support kitti format with velocity ground truth and sweeps?

kentang-mit, 2023-03-06:
Hi @Qjizhi,

We are still working on it and the Waymo configs are expected to come out in the next release. We have a custom format for Waymo data representation which is more similar to nuScenes. 

Best,
Haotian

Qjizhi, 2023-03-06:
> Hi @Qjizhi,
> 
> We are still working on it and the Waymo configs are expected to come out in the next release. We have a custom format for Waymo data representation which is more similar to nuScenes.
> 
> Best, Haotian

@kentang-mit Thank you, good to know

closed 2023-03-30 by kentang-mit