mit-han-lab/bevfusion #262: KeyError: 'Unrecognized norm type SubMConv3d'

wanghangege, 2022-12-09:
Can you help me solve the problem, mmcv or spconv ? I can not search the similar problem...  
File "/home/seu2/anaconda3/envs/bevf/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/home/seu2/bbwang/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 75, in __init__
    self.conv_input = make_sparse_convmodule(
  File "/home/seu2/bbwang/bevfusion/mmdet3d/ops/sparse_block.py", line 157, in make_sparse_convmodule
    build_conv_layer(
  File "/home/seu2/anaconda3/envs/bevf/lib/python3.8/site-packages/mmcv/cnn/bricks/conv.py", line 38, in build_conv_layer
    raise KeyError(f'Unrecognized norm type {layer_type}')
KeyError: 'Unrecognized norm type SubMConv3d'

kentang-mit, 2022-12-10:
My suggestion will be to reinstall `mmcv==1.4.0`, `mmcv-full==1.4.0` and run

```bash
python setup.py develop
```

under the root folder.

wanghangege, 2022-12-12:
Thanks for your reply, I will try it soon, if it works, I will feedback to you.
Besides, I have achieve the train and visiualize in 3090 gpu, the spconv is spconv (by the "spconv.__version__"), because to I  have only one GPU, so I did not complete the training process. Thank you again.

kentang-mit, 2022-12-12:
Glad to hear that. We also have an internal version based on spconv 2.x and I can confirm that there should not be major performance difference if you use that version for all training experiments.

kentang-mit, 2022-12-27:
Closed due to inactivity.

closed 2022-12-27 by kentang-mit