mit-han-lab/bevfusion #263: How to train a multitask model?

ycwang1997, 2022-12-09:
Hello,

Thanks for your great work! I want to train a camera-only multitask model for det and seg tasks. Could you release the multitask configs and pretrained models?
Thank you.

kentang-mit, 2022-12-10:
See https://github.com/mit-han-lab/bevfusion/issues/252. We do not release the multi-task configs and pretrained models for now because none of them achieve as good results as single-task models. You are welcomed to do some explorations on this direction.

ycwang1997, 2022-12-12:
Thanks for your reply.
I specified the configuration in `bevfusion/configs/nuscenes/det/default.yaml`, like
```
model:
  type: BEVFusion
  encoders: null
  fuser: null
  heads:
    map: 
      type: BEVSegmentationHead
      in_channels: 256
      grid_transform:
        input_scope: [[-51.2, 51.2, 0.8], [-51.2, 51.2, 0.8]]
        output_scope: [[-50, 50, 0.5], [-50, 50, 0.5]]
      classes: ${map_classes}
      loss: focal
```
I use 4 GPUs(A6000) to train the model by runing the follow:
`torchpack dist-run -np 4 python tools/train.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth`

However, the losses became **nan** suddenly after training 5 epochs.
[image]

I have tried to train again, but the same problem occured after training 4 epochs.
Could you please help me with the problem?
Thank you.

kentang-mit, 2022-12-13:
I probably haven't tried camera-only multi-task model. But according to my experience on multi-modal multi-task models, it is beneficial to separate the BEV encoders for different tasks (see Table 5 in our arXiv version). There are other things you can try, for example adjusting the loss ratio for detection and segmentation, or pretrain the detection model and then jointly finetune det+seg.

zehaoj, 2023-03-16:
Hi,
Thank you so much for this great work, clear paper and code structure!

I'm just wondering if it's possible to share the config for multitask model? I'm currently doing my master's thesis about multi-task learning and found your work very interesting. It would be really helpful if you could provide that. My email is zehaoj@kth.se

Thanks!!

kentang-mit, 2023-03-23:
@zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.

closed 2023-03-23 by kentang-mit

vehxianfish, 2023-03-30:
> @zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.

Thanks for your wonderful work! Can you send the multi-task training config files to me? Thanks a lot. My email is s_xu@foxmail.com

ycwang1997, 2023-03-31:
> @zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.

Thanks for your suggestion. I have completed the camera-only multi-task training. Actually, I'm doing my master's research about multi-task, too.
It is helpful if you can provide your multi-task config for multi-sensor to let me compare the difference. My email is ycwang1223@gmail.com. Thanks!

ldtho, 2023-08-13:
Hi guys, can I ask for multi-task config as well? my email is ldtho.97@gmail.com 
I'm doing research on multi-task learning and found this repo really interesting, thank you so much for your help.

ChunyuFeng, 2023-09-26:
> @zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.

It's a great work! Could you please send me the multi-task training config files? Thank you very much! My email is tj.chunyufeng@gmail.com

AvinsWang, 2023-09-28:
> @zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.

Cool work! I want to train the multi-modal multi-task model too. I would be grateful if you sent me the config. My email is avinswang@gmail.com. Thanks.

heli223, 2024-01-17:
I also want to train the multi-modal multi-task model too. Can anyone send the config and related codes to me? My email is 1632226712@qq.com. Many thanks!

think-twice-1218, 2024-02-27:
> @zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.

Thanks for your excellent job!  I want to train the multi-modal multi-task model too. Could you send me a copy of the config and codes? My email is 1106205197@qq.com , Thanks.

autumn-wong, 2024-03-08:
> Hi, @kentang-mit Thank you so much for this great work, clear paper and code structure!
> 
> I'm just wondering if it's possible to share the config for multitask model? I'm currently doing my master's thesis about multi-task learning and found your work very interesting. It would be really helpful if you could provide that. My email is [zehaoj@kth.se](mailto:zehaoj@kth.se)
> 
> Thanks!!

你好，请问您那有多任务的config文件吗，可以分享一下吗？我的邮箱是2085985523@qq.com
助力每一个毕业梦想，非常感谢！！！

vehxianfish, 2024-03-19:
> Hi, @kentang-mit Thank you so much for this great work, clear paper and code structure!
>
> I'm just wondering if it's possible to share the config for multitask model? I'm currently doing my master's [...]

@zehaoj, Can you send multi-task config to me, tks a lot. My email is s_xu@foxmail.com

vehxianfish, 2024-03-19:
> > @zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.
> 
> Thanks for your wonderful work! Can you send the multi-task training config files to me? Thanks a lot. My email is [s_xu@foxmail.com](mailto:s_xu@foxmail.com)

@kentang-mit. Thanks for your wonderful work! Can you send the multi-task training config files to me? Thanks a lot. My email is [s_xu@foxmail.com](mailto:s_xu@foxmail.com)

shupinghu, 2024-05-29:
> @zehaoj, I've sent you an Email about the config. I'm closing this issue since it is resolved.

Hello, I'm currently doing some research about multi-task learning and found your work very interesting. But it is difficult for me to train a multitask model. It would be really helpful if you could provide that.I wander If it's possible to share the config for multitask model for me? My email is hushp3@qq.com.

Thanks!!