mit-han-lab/bevfusion #265: Do the lidar-only branch need to load a pre-trained voxel model during training to improve accuracy or speed up convergence?

BHC1205, 2022-12-12:
As the title suggests.

kentang-mit, 2022-12-12:
I think loading the pretrained model will both improve accuracy and speed up the convergence of a multi-modal model.

BHC1205, 2023-01-17:
I have a pre-training model for lidar branch, and I want to add a small part of the network structure to it, should I freeze Lidar encoder part and train transformer part only?
Can you give some advice(such as lr)? lots of Thanks.

kentang-mit, 2023-01-17:
In this case, my suggestion will be to finetune the entire fusion model end-to-end with your randomly initialized new module. Note that re-doing the LiDAR pretraining with your new module is usually unnecessary according to my experience. If you freeze the LiDAR encoder I guess it can also work, but the performance might be worse.

closed 2023-03-30 by kentang-mit