mit-han-lab/bevfusion #269: measure inference and training time

alchemz, 2022-12-15:
Dear author, following you recommendation, I have added timer for the forward function part of bevfusion.py.
[in forward_single(), before bevfusion.py#L225](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/fusion_models/bevfusion.py#L225)
```
torch.cuda.synchronize()
t_start = time.time()
features = []
```
[After bevfusion.py#L304 ](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/fusion_models/bevfusion.py#L304)
```
torch.cuda.synchronize()
t_forward_test_end = time.time()
print('forward_test, ', t_end_test - t_start)
return outputs
```

Then I got the prints as the following, the unit for time.time() is **milliseconds**. I wonder how do you get the overall **~130ms** inference time? @kentang-mit. Is the unit of time.time() **second** instead?
```
forward_test,  0.18337655067443848
forward_test,  0.17694544792175293
forward_test,  0.18417620658874512
forward_test,  0.18697500228881836
forward_test,  0.19140982627868652
forward_test,  0.2163400650024414
forward_test,  0.20026397705078125
forward_test,  0.1974184513092041
forward_test,  0.2000215053558349
....
```

alchemz, 2022-12-15:
It seems like the unit for t_end_test - t_start should be s.  Then 0.18337655067443848 is 183.3ms

kentang-mit, 2022-12-15:
We obtain the latency in our paper by summing up the runtime of different components in the model (e.g. LiDAR encoder, camera encoder, view transformer, BEV network, etc.) The major part that is excluded is the voxelization time, because this operation is trivially parallelizable and the current implementation in mmdet3d is not optimal.