mit-han-lab/bevfusion #271: The problem about evaluating and training

IcyMing, 2022-12-17:
Hello~Thanks for your perfect work! But I met some problems when evaluating and training...
For example, when I used the command as below.
`torchpack dist-run -np 2 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`
Then ,the errors are as follows:
[image]

Some important packages in environment：
```
python==3.8
torch==1.9.1+cu111
torchvision==0.10.1+cu111
torchaudio==0.9.1
mmcv-full==1.4.0
mmdet==2.20.0
tqdm==4.64.1 
torchpack==0.3.1
nuscenes-devkit==1.1.9
mpi4py==3.1.4
ninja==1.11.1
numba==0.56.4
setuptools==58.0.4 
 ```

kentang-mit, 2022-12-18:
I don't think the current error message is very informative. Could you please give me the specific error you met with (e.g. some people may see something like "no kernel image available for execution")? By the way, if you cannot see any informative error messages, try adding a `-v` flag to the command will be a good idea.

IcyMing, 2022-12-18:
> I don't think the current stdout message is very informative. What you show me are warnings instead of errors. Could you please give me the specific error you met with (e.g. some people may see something like "no kernel image available for execution")? By the way, if you cannot see any informative error messages, try adding a `-v` flag to the command will be a good idea.

Thank you for your kind reply! I tried adding a `-v` flag to the command, and the errors are as follows:
[image]

kentang-mit, 2022-12-19:
Thanks for the update. It seems that it is a problem related to the `MultiPolygon` class. May I ask what is the version of your `shapely` library? We use 1.8.0 in our experiments.

rylynchen, 2023-01-10:
Training got error with shapely==2.0.0, then change to 1.8.0, it work

IcyMing, 2023-01-13:
> 

Thank you very much! It works!

closed 2023-01-13 by IcyMing