mit-han-lab/bevfusion #274: How to choose proper head for different modalities?

sunnyHelen, 2022-12-24:
Hi, thanks a lot for sharing your great work. 
I noticed that you use the centerhead for the camera modality and the transfusion head for the lidar and fusion ones. Is there any observation or conclusion about how to choose proper heads for different modalities? Did you try the other versions? E.g., the transfusion head for the camera modality and the centerhead for the fusion modality...

kentang-mit, 2022-12-24:
Hi @sunnyHelen,

Good question! The short answer is that centerhead usually works for all the tasks and datasets. If you have a new task or new dataset, I would recommend you to try out centerhead first. For transfusion head, we used it on nuScenes detection because it achieved higher performance compared with centerhead. To use it on other datasets, it might require some efforts to tune the parameters (e.g. number of proposals). By the way, the transfusion head is similar to a two-stage detection head. In this case, it would require the first stage predictions (in its case, the center heatmap predictions) to be relatively accurate. So it might be challenging (but still possible) to apply transfusion head to camera-only detectors.

Best,
Haotian

sunnyHelen, 2022-12-24:
Thanks for your reply, Haotian. Being inspired a lot~

closed 2022-12-24 by sunnyHelen