mit-han-lab/bevfusion #275: AP 0 for 3d object detection task

alchemz, 2022-12-25:
Per-class results:
Object Class    AP      ATE     ASE     AOE     AVE     AAE
car     		0.027   1.138   0.177   0.631   0.210   0.031
truck   		0.000   1.067   0.332   0.135   0.057   0.000
bus     		0.000   1.000   1.000   1.000   1.000   1.000
trailer 		0.000   1.000   1.000   1.000   1.000   1.000
const_vehicle  0.000   1.000   1.000   1.000   1.000   1.000
pedestrian       0.009   1.285   0.284   1.291   0.656   0.492
motorcycle      0.015   1.432   0.312   1.539   0.104   0.075
bicycle 0.000   1.000   1.000   1.000   1.000   1.000
traffic_cone    0.000   1.000   1.000   nan     nan     nan
barrier 0.000   1.000   1.000   1.000   nan     nan

{'pts_bbox_NuScenes/car_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/car_AP_dist_1.0': 0.0003, 'pts_bbox_NuScenes/car_AP_dist_2.0': 0.0284, 'pts_bbox_NuScenes/car_AP_dist_4.0': 0.0788, 'pts_bbox_NuScenes/car_trans_err': 1.1382, 'pts_bbox_NuScenes/car_scale_err': 0.1768, 'pts_bbox_NuScenes/car_orient_err': 0.6307, 'pts_bbox_NuScenes/car_vel_err': 0.2097, 'pts_bbox_NuScenes/car_attr_err': 0.0315, 'pts_bbox_NuScenes/mATE': 1.0922, 'pts_bbox_NuScenes/mASE': 0.7104, 'pts_bbox_NuScenes/mAOE': 0.9551, 'pts_bbox_NuScenes/mAVE': 0.6283, 'pts_bbox_NuScenes/mAAE': 0.5749, 'pts_bbox_NuScenes/truck_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/truck_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/truck_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/truck_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/truck_trans_err':1.0665, 'pts_bbox_NuScenes/truck_scale_err': 0.3317, 'pts_bbox_NuScenes/truck_orient_err': 0.1347, 'pts_bbox_NuScenes/truck_vel_err': 0.057, 'pts_bbox_NuScenes/truck_attr_err': 0.0, 'pts_bbox_NuScenes/construction_vehicle_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/construction_vehicle_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/construction_vehicle_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/construction_vehicle_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/construction_vehicle_trans_err': 1.0, 'pts_bbox_NuScenes/construction_vehicle_scale_err': 1.0, 'pts_bbox_NuScenes/construction_vehicle_orient_err': 1.0, 'pts_bbox_NuScenes/construction_vehicle_vel_err': 1.0,'pts_bbox_NuScenes/construction_vehicle_attr_err': 1.0, 'pts_bbox_NuScenes/bus_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/bus_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/bus_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/bus_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/bus_trans_err': 1.0, 'pts_bbox_NuScenes/bus_scale_err': 1.0, 'pts_bbox_NuScenes/bus_orient_err': 1.0, 'pts_bbox_NuScenes/bus_vel_err': 1.0, 'pts_bbox_NuScenes/bus_attr_err': 1.0, 'pts_bbox_NuScenes/trailer_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/trailer_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/trailer_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/trailer_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/trailer_trans_err':1.0, 'pts_bbox_NuScenes/trailer_scale_err': 1.0, 'pts_bbox_NuScenes/trailer_orient_err': 1.0, 'pts_bbox_NuScenes/trailer_vel_err': 1.0, 'pts_bbox_NuScenes/trailer_attr_err': 1.0, 'pts_bbox_NuScenes/barrier_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/barrier_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/barrier_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/barrier_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/barrier_trans_err': 1.0, 'pts_bbox_NuScenes/barrier_scale_err': 1.0, 'pts_bbox_NuScenes/barrier_orient_err': 1.0, 'pts_bbox_NuScenes/barrier_vel_err': nan, 'pts_bbox_NuScenes/barrier_attr_err': nan, 'pts_bbox_NuScenes/motorcycle_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/motorcycle_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/motorcycle_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/motorcycle_AP_dist_4.0': 0.0609, 'pts_bbox_NuScenes/motorcycle_trans_err': 1.4322, 'pts_bbox_NuScenes/motorcycle_scale_err': 0.3118, 'pts_bbox_NuScenes/motorcycle_orient_err': 1.5394, 'pts_bbox_NuScenes/motorcycle_vel_err': 0.1036, 'pts_bbox_NuScenes/motorcycle_attr_err': 0.0754, 'pts_bbox_NuScenes/bicycle_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/bicycle_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/bicycle_AP_dist_2.0':0.0, 'pts_bbox_NuScenes/bicycle_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/bicycle_trans_err': 1.0, 'pts_bbox_NuScenes/bicycle_scale_err': 1.0, 'pts_bbox_NuScenes/bicycle_orient_err': 1.0, 'pts_bbox_NuScenes/bicycle_vel_err': 1.0, 'pts_bbox_NuScenes/bicycle_attr_err': 1.0, 'pts_bbox_NuScenes/pedestrian_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/pedestrian_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/pedestrian_AP_dist_2.0': 0.0008, 'pts_bbox_NuScenes/pedestrian_AP_dist_4.0': 0.0338, 'pts_bbox_NuScenes/pedestrian_trans_err': 1.2851, 'pts_bbox_NuScenes/pedestrian_scale_err': 0.2839, 'pts_bbox_NuScenes/pedestrian_orient_err': 1.2908, 'pts_bbox_NuScenes/pedestrian_vel_err': 0.6561, 'pts_bbox_NuScenes/pedestrian_attr_err': 0.492, 'pts_bbox_NuScenes/traffic_cone_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/traffic_cone_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/traffic_cone_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/traffic_cone_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/traffic_cone_trans_err': 1.0, 'pts_bbox_NuScenes/traffic_cone_scale_err': 1.0, 'pts_bbox_NuScenes/traffic_cone_orient_err': nan, 'pts_bbox_NuScenes/traffic_cone_vel_err': nan, 'pts_bbox_NuScenes/traffic_cone_attr_err': nan, 'pts_bbox_NuScenes/NDS': 0.11567509610720243, 'pts_bbox_NuScenes/mAP': 0.005075794611831504}

alchemz, 2022-12-25:
I am trying to replace voxelnet with pointpillar for lidar backbone, used voxel size 0.16, but got 0 AP for 3d object detection task. @kentang-mit Dear Author, have you experimented with pointpillar as lidar backbone? And could you share your config is applicable?

kentang-mit, 2022-12-27:
Yes we did, but we haven't done experiments on sensor fusion with the PointPillars backbone. A **LiDAR-only** PointPillars should be able to achieve 54-55 mAP on nuScenes. You could refer to mmdet3d for official configurations and remember to turn on `gt_paste_stop_epoch: 15` will help you get these numbers.

alchemz, 2022-12-27:
@kentang-mit Than you. I will try it and let you know.

alchemz, 2023-01-08:
@kentang-mit Dear author, I noticed that you have also released the [pointpillar.yaml](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/det/transfusion/secfpn/lidar/pointpillars.yaml) for lidar only benchmark , and would like to replicate the experiment using this config. But I am running to the below errors 
```
File "/bevfusion/mmdet3d/models/utils/transformer.py", line 69, in with_pos_embed
return tensor if pos_embed is None else tensor + pos_embed
RuntimeError: The size of tensor a (16384) must match the size of tensor b (65536) at non-singleton dimension 0
```
Could you share elaborate what could be the cause?

alchemz, 2023-01-08:
never mind. I have figured it out. Mistakenly changed grid size previoulsy. Now it trains without any errors.

closed 2023-01-08 by alchemz

HenryZhangJianhe, 2023-09-20:
> never mind. I have figured it out. Mistakenly changed grid size previoulsy. Now it trains without any errors.

I use default config but I met same error, could you tell me how to change config?

odroidodroid, 2024-01-05:
@alchemz  I have the same issue. could you tell me how to fix the errors?