mit-han-lab/bevfusion #278: point cloud range update

RG2806, 2023-01-03:
hey,
Thanks for open sourcing this work
I wanted to know if i want to detect objects only in front of lidar, apart changing this parameter
https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/default.yaml#L9

what else i would have to change?
for ex: I want to use  [0.0, -54.0, -5.0, 108.0, 54.0, 3.0] as my range
How will this change affect the vtransform xbounds or voxel grid size?

thanks

kentang-mit, 2023-01-03:
Yes, I think you also need to change the vtransform xbounds but you do not need to change the voxel grid sizes. For the final image size, it will be better to scale [image_size](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/configs/nuscenes/default.yaml#L11) and [2D augmentation scales](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/configs/nuscenes/default.yaml#L14) accordingly. Just make sure the desired image size approximately lies in the middle of your augmentation interval. By the way, it is recommended to make sure that the final image size is divisible by 32.

RG2806, 2023-01-04:
hey @kentang-mit,
can you tell why have you defined point cloud ranges in different places, as here,
https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/default.yaml#L9
https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/det/transfusion/secfpn/camera%2Blidar/swint_v0p075/default.yaml#L2

RG2806, 2023-01-04:
thanks for the clarification.
One other thing, i was trying to train the lidar only model with pointpillars as backbone. And I received this error.
This is the trace
File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/apis/train.py", line 140, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/runner/epoch_based_runner.py", line 253, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/runner/epoch_based_runner.py", line 315, in train
    super().train(data_loader, **kwargs)
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/runner/epoch_based_runner.py", line 173, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/runner/epoch_based_runner.py", line 152, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/home/predev/.local/lib/python3.8/site-packages/mmcv/parallel/distributed.py", line 52, in train_step
    output = self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/home/predev/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/predev/.local/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/models/fusion_models/bevfusion.py", line 189, in forward
    outputs = self.forward_single(
  File "/home/predev/.local/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/models/fusion_models/bevfusion.py", line 272, in forward_single
    pred_dict = head(x, metas)
  File "/home/predev/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/models/heads/bbox/transfusion.py", line 353, in forward
    res = multi_apply(self.forward_single, feats, [None], [metas])
  File "/home/predev/.local/lib/python3.8/site-packages/mmdet/core/utils/misc.py", line 30, in multi_apply
    return tuple(map(list, zip(*map_results)))
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/models/heads/bbox/transfusion.py", line 306, in forward_single
    query_feat = self.decoder[i](
  File "/home/predev/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/models/utils/transformer.py", line 100, in forward
    key=self.with_pos_embed(key, key_pos_embed),
  File "/home/predev/repos/rgupta38/bevfusion-git/mmdet3d/models/utils/transformer.py", line 69, in with_pos_embed
    return tensor if pos_embed is None else tensor + pos_embed
RuntimeError: The size of tensor a (16384) must match the size of tensor b (65536) at non-singleton dimension 0

this is the logfile
[20230104_194735.txt](https://github.com/mit-han-lab/bevfusion/files/10347053/20230104_194735.txt)

can you help here @kentang-mit

kentang-mit, 2023-01-05:
Hi @RG2806,

I never tried PointPillars with a transfusion head. However It could work with the centerhead. The configuration for the LiDAR part looks like this:

```yaml
voxel_size: [0.2, 0.2, 8]

model:
  encoders:
    lidar:
      voxelize_reduce: false
      voxelize:
        max_num_points: 20
        point_cloud_range: ${point_cloud_range}
        voxel_size: ${voxel_size}
        max_voxels: [30000, 60000]
      backbone:
        type: PointPillarsEncoder
        pts_voxel_encoder:
          type: PillarFeatureNet
          in_channels: 5
          feat_channels: [64, 64]
          with_distance: false
          point_cloud_range: ${point_cloud_range}
          voxel_size: ${voxel_size}
          norm_cfg:
            type: BN1d
            eps: 1.0e-3
            momentum: 0.01
        pts_middle_encoder:
          type: PointPillarsScatter
          in_channels: 64
          output_shape: [512, 512]
  decoder:
    backbone:
      in_channels: 64
      out_channels: [64, 128, 256]
      layer_nums: [3, 5, 5]
      layer_strides: [2, 2, 2]
    neck:
      in_channels: [64, 128, 256]
      out_channels: [128, 128, 128]
      upsample_strides: [0.5, 1, 2]
  heads:
    object:
      in_channels: 384
      train_cfg:
        grid_size: [512, 512, 1]
        out_size_factor: 4
      test_cfg:
        out_size_factor: 4
      bbox_coder:
        out_size_factor: 4
```

Best,
Haotian

RG2806, 2023-01-05:
Hey @kentang-mit 
Thanks for the clarification

closed 2023-01-05 by RG2806

hollow-503, 2023-02-24:
> multiple of 32

@kentang-mit 
Hi! Could you tell me why the range / voxel size is recommended to be a multiple of 32 ?