mit-han-lab/bevfusion #280: question about origin center

U201310646, 2023-01-06:
Hi,
the same question but i wanna ask if this change  will have an impact on the lidar's data augmentation? 
> https://github.com/mit-han-lab/bevfusion/blob/6908c57b7800268a9cdf3efeacbfb886d6077d6f/mmdet3d/datasets/nuscenes_dataset.py#L314-L317
origin set to (0.5,0.5,0), the following code wont be executed, the center is still geometric center of b-box
> https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/core/bbox/structures/base_box3d.py#L61-L64      

but i noticed that  the origin here (0.5,0.5,0) in points_in_rbbox function, the center is considered as center of bottom surface to caculate the corners. Is this inconsistent?
> https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/core/bbox/box_np_ops.py#L430

_Originally posted by @U201310646 in https://github.com/mit-han-lab/bevfusion/issues/128#issuecomment-1373207594_

kentang-mit, 2023-01-11:
Hi @U201310646,

Thank you for this issue and sorry for the late response. I was dealing with some other projects these days. After reading the code you pasted, I feel that if the code snippet in `core/bbox/structures/base_box3d.py` is not executed, the origin should be the center of the bottom surface, which matches with the default value in `core/bbox/box_np_ops.py`. But I would have a deeper look at this part and do some rigorous check on whether the GT databases generated by our code is indeed correct. 

Best,
Haotian

U201310646, 2023-01-12:
Thanks, looking forward to your reply

U201310646, 2023-01-12:
The x,y,z annotation in Nuscenes is the geometric center. But in LiDARInstance3DBoxes with origin(0.5,0.5,0) the code snippet in _core/bbox/structures/base_box3d.py_ will do nothing for origin. If origin set to (0.5,0.5,0.5), then the origin will be converted to the center of bottom surface, which then matched wth the default args in _core/bbox/box_np_ops.py_. Please correct me if i missuderstood.

kentang-mit, 2023-01-12:
I think your understanding is correct. I did some check on the GT database generated with the released code and found it to be different from what we generated one year ago (and used in the experiments). I'm running more experiments to verify the impact.

U201310646, 2023-01-13:
thanks for your reply!

closed 2023-02-23 by U201310646

YongYongLi123, 2024-02-02:
> which then matched wth the default args in _core/bbox/box_np_ops.py_. Please correct me if i missuderstood.

I also have the same doubts, and I think your understanding is correct