mit-han-lab/bevfusion #283: CUDA error: no kernel image

sunnyHelen, 2023-01-11:
Hi, Haotian, I also encountered a problem "RuntimeError: CUDA error: no kernel image is available for execution on the device".

I searched the other issues and tried to remove all kinds of versions of the four lines like "-gencode=arch=compute_75,code=sm_75". But it seems I didn't find the right one for my environment.
Here is my environment:
GTX 3090
CUDA 10.1
torch==1.8.1+cu101 
torchvision==0.9.1+cu101 
torchaudio==0.8.1
Do you know the right compute and sm version for it?

kentang-mit, 2023-01-11:
Hi @sunnyHelen,

For your device the sm version should be sm_86 and compute_86. Actually I would recommend you to upgrade the CUDA version to 11.0 and PyTorch version to 1.9.x or 1.10.x to better align with our experiment settings.

Best,
Haotian

sunnyHelen, 2023-01-30:
Thank you very much. Solved.

closed 2023-01-30 by sunnyHelen