mit-han-lab/bevfusion #285: about coordinates system matching

alchemz, 2023-01-12:
Dear author, as shown in extract_camera_features, there is camera to lidar transformation at camera feature extraction steps. 
https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/fusion_models/bevfusion.py#L88

As it is known the nuscenes GT is in world coordinates. **Could you confirm if both camera and lidar are aligned in lidar coordinates==equilavent to world coordinates in GT?**

In addition, when I print out the min and max bound of `img_feats` and `pts_feats`, it is shown as below, does the (0.0, 41.59375) bound representing the world/lidar coordinates, as the origin is from top left corner of the map.
```
img_feats.min 0.0 pts_feats.min 0.0
img_feats.max 41.59375 pts_feats.max 28.71875
```
Lastly, when I print out coors from [extract_lidar_features]. It shows `pts coors tensor([[  0, 680, 700,  14],
        [  0, 564, 760,  26],
        [  0, 728, 835,  16],
        ...,
        [  3, 273, 894,  34],
        [  3, 821, 891,  24],
        [  3, 460, 685,  37]], device='cuda:0', dtype=torch.int32)`. **Is the [  3, 460, 685,  37], representing x, y, z, ?**
https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/fusion_models/bevfusion.py#L129

Thanks for your help!

kentang-mit, 2023-01-12:
Hi @alchemz,

Thanks for your interest. For your questions,

1. Both camera and LiDAR features are aligned in LiDAR-centric coordinate systems. The GT bounding boxes are in the world coordinate system, and we convert them to the LiDAR-centric coordinate system for training purposes.
2. If you are talking about the range of features, I don't think they directly correlates with the range of coordinates. Would you mind telling me what is the range for the `img_feats` and `pts_feats` you mentioned in the issue?
3. It is `batch_index, x, y, z`.

Best,
Haotian

alchemz, 2023-01-12:
@kentang-mit Thanks for your prompt response.

1. followup: 
  a. what is the benefits of training using lidar-centric vs ego-centirc vs world coordinates?
  b.  as GT in world coor, training in lidar coor, where does it in code to transform the predicted bbox to world?

2. followup:
  a. by range for the `img_feats` and `pts_feats`, do you mean `point_cloud_range`?
  b. how does the feature range indirectly related to coordinates range?
  c. for reference:
    * `point_cloud_range`=[-51.2, -51.2, -5.0, 51.2, 51.2, 3.0])  
    * 1 example `img_feats` range: (img_feats.min, img_feats.max) -> (0, 41.59375)
    * 1 example `pts_feats` range: (pts_feats.min, pts_feats.max) -> (0, 28.71875)

kentang-mit, 2023-01-12:
Hi @alchemz,

For 1, training using LiDAR-centric and Ego-centric coordinate systems should give you roughly the same performance. World coordinate system is a bit tricky seems it is very hard for you to define the `point_cloud_range`. Ultimately some normalization is needed.

For 2, actually I'm not very sure about the meaning of `img_feats` and `pts_feats` in the context of your question. If it means features extracted by camera and LiDAR encoders, then their ranges do not directly correspond to the range of coordinates (since these are high dimensional features). 

Best,
Haotian

alchemz, 2023-01-13:
I got the img_feats and pts_feats extracted from the bevfusion.py using extract_camera_features, extract_lidar_features, and thought the bound of the features could indicate if it is ego or lidar coordinates. But its apparently wrong. Thanks Haotian for the clarification, it is very helpful.

I am interested to convert both image and lidar to ego coordinates, and check the experiments performance. To do so
1. convert image to lidar coor to ego coor: 
  - Todo: cam2ego in [LoadMultiViewImageFromFiles](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/datasets/pipelines/loading.py#L19)
2. convert lidar pts to ego coord: 
  - Todo: lidar2ego in [LoadPointsFromMultiSweeps](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/datasets/pipelines/loading.py#L84)
3. convert gt to ego: 
  - Todo: gt2ego in https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/datasets/nuscenes_dataset.py#L209-L231

@kentang-mit  Could you help confirm if the 3 things are needed to do for lidar centric to ego centric conversion?

I am not sure, as there are also. To convert to ego, does it have to be **box_type_3d**: Camera, and **coord_type**: CAMERA?

kentang-mit, 2023-01-13:
Hi @alchemz,

Your understanding is correct. But please make sure that the ego coordinate systems for camera and LiDAR are the same coordinate system. Previously we encountered some problems here (basically camera_to_ego x ego_to_lidar != camera_to_lidar, you need to double check it). By the way, for box_type and coord_type, you can still use `LiDAR`.

Best,
Haotian

alchemz, 2023-01-15:
@kentang-mit Thank you Haotian for the kind reminder to double checking coordinate transformation. I need to convert the loaded points to ego frame.
Could you confirm if the below code are doing `a. point2lidar` or `b.point2ego` transformation? 

https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/datasets/pipelines/loading.py#L220-L226

If a. then there is no addtional step needed for points/sweep transformation to ego

if b. to transform the loaded points to ego, after the ln 226, will it be **point2ego = lidar2ego @ point2lidar ?**

alchemz, 2023-01-15:
```
sensor2lidar = create_pose_matrix(translation=sweep['sensor2lidar_translation'],
                                        rotation_matrix=sweep['sensor2lidar_rotation'])
lidar2ego = create_pose_matrix(translation=sweep['lidar2ego_translation'],
                                    rotation_matrix=sweep['lidar2ego_rotation'])
poin2ego = lidar2ego @ sensor2lidar

points_sweep[:, :3] = points_sweep[:, :3] @ poin2ego[:3, :3].T 
points_sweep[:, :3] += + poin2ego[:3, 3]
```
@kentang-mit This is what I have now for point2ego transformation.

kentang-mit, 2023-01-17:
Hi @alchemz,

For your first thread, the transformation is point2lidar. The purpose of lines 220-226 in the file you referred to is to convert all LiDAR samples in the same sweep to the same coordinate system such that these LiDAR scans could be superimposed. For the second thread, I think what you have done is correct for point->ego, but as I mentioned previously, you may want to check whether camera2ego x ego2lidar = camera2lidar later when you have camera2ego. 

Best,
Haotian

closed 2023-02-10 by alchemz

lilkeker, 2023-05-15:
@alchemz Hello,recently I've been interested in implementing BevFusion in the Ego system. Your discussions and responses with Haotian have been incredibly helpful to me, and I greatly appreciate it! However, I'm facing an issue with the map generated by the lidar model (Transfusion-Lidar) I implemented in the Ego system. After training for one epoch, the map appears to be lower than expected. I'm not sure if there are any issues with the coordinate  conversion process or if I have missed something. BTW, I have visualized it, and it looks normal.

[image]
关于lidar2ego 的定义如下：

alchemz, 2023-06-01:
> @alchemz Hello, recently I also implemented BevFusion in the Ego system. Your discussions and responses with Haotian have been incredibly helpful to me, and I greatly appreciate it! However, I'm facing an issue with the map generated by the lidar model (Transfusion-Lidar) I implemented in the Ego system. After training for 15 epoch, the mAP appears to be lower than expected(mAP: 0.3240 NDS: 0.4040). I'm not sure if there are any issues with the coordinate conversion process or if I have missed something. BTW, I have visualized it, and it looks normal. I defined Coordinate_change and it happens after ObjectSample, below is my code，I would be very grateful if you could help me in your busy schedule! [image] [image] The definition of lidar2ego is as follows: lidar2ego = np.eye(4, dtype=np.float32) lidar2ego[:3, :3] = Quaternion(info['lidar2ego_rotation']).rotation_matrix lidar2ego[:3, 3] = info['lidar2ego_translation'] lidar2ego = torch.from_numpy(lidar2ego)

Hi there,

Could you try to comment out dbsampler and verify if model performance are similar between ego and lidar coordinates? Without dbsampleer, the lidar model should be around 60 NDS.