mit-han-lab/bevfusion #286: tensorrt aceleration?

jinfagang, 2023-01-13:
does bevfusion able to inference with tensorrt?

kentang-mit, 2023-01-13:
Hi @jinfagang,

We are still working on TensorRT support. It requires us to implement sparse convolution layers and the bev_pool kernel as TensorRT plugins.

Best,
Haotian

kentang-mit, 2023-03-30:
Closed due to inactivity. Please feel free to reopen if you feel it necessary.

closed 2023-03-30 by kentang-mit

philferriere, 2023-03-30:
Hi @kentang-mit ,

Thank you again for your hard work on BEVFusion. This is an incredible multi-modal model you and your team have put together!

I noticed you close this issue, due to inactivity. Do you happen to have an update regarding @jinfagang 's original query? Are you working on the plugins or is this work that is not being  considered anymore?

Thanks for clarifying!

Best, -- Phil