mit-han-lab/bevfusion #287: The output shape of voxelization

JingweiZhang12, 2023-01-17:
Thanks for your work!
I wonder that the motivation for this modification: https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/ops/voxel/voxelize.py#L119. 
If I use the original implementation of mmdet3d and modify the corresponding sparse shape, will it have an influence on the performance?

kentang-mit, 2023-01-17:
I think the speed will be exactly the same. But please be extra careful about the correctness after doing such a modification. For example, the (x, y) coordinates of the object centers might need to be transposed. See code [here](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/heads/bbox/centerpoint.py#L510-557) and [here](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/heads/bbox/transfusion.py#L568) for more details.

JingweiZhang12, 2023-01-18:
Thanks for your timely reply. I will have a try.

closed 2023-03-30 by kentang-mit